TriForceX/MiyooCFW #4: Invalid command spiflash-write

curlyj2fo, 2019-03-17:
I am not sure what I am doing wrong but when I get to step 3.3 6. I get a message saying "Invalid command spiflash-write"

I am using a clean boot of Ubuntu 18.04 and read that these would help.
$ sudo apt install make
$ sudo apt-get install build-essential
$ sudo apt-get install pkg-config
$ sudo apt-get install libz-dev 
$ sudo apt-get install libusb-1.0-0-dev

I still the same message "Invalid command spiflash-write"

curlyj2fo, 2019-03-17:
On step 3.2-4 I had to do a slightly different command.
I read that after 'make clean && make' I needed to do 'make install'
But it gave me invalid permissions so I did 'sudo make install'
Then I proceeded with the steps as usual and it worked.

closed 2019-03-17 by curlyj2fo

TriForceX, 2019-04-14:
Thanks we will update the guide with that advice.