TriForceX/MiyooCFW #10: Neogeo Pocket

dilbirin, 2019-04-11:
Is race! emulator possibble?

TriForceX, 2019-04-12:
Yes it is, and the port for bittboy is in the to-do list.

dilbirin, 2019-04-12:
Well.And another request :) Is there j2me emulator in your to do list? Thank you for your interest.

deadl0ck, 2019-08-01:
Delighted to read this @TriForceX (just bought a PocketGo) - has there been much progress since you posted about this ?
Thanks !

gameblabla, 2019-08-01:
I'm working on my own Neo geo pocket emulator myself for the Retrostone, RS-97, Bittboy... and it turns out it runs too slowly unless you overclock it up to 798 Mhz or so.
Obviously i'm not going to release in such a state.

deadl0ck, 2019-08-01:
Interesting @gameblabla - maybe if you release it others could help improve the performance? One way or another it's great to see people working on this

gameblabla, 2019-08-01:
Yeah i've been thinking to release it as is though i'm also working on a proper CFW for the bittboy... (So i would think i would rather release it for that)

Gave a preliminarily of VBA Next to someone else and he confirmed my findings. (It would only run fast enough when overclock at 798 too)

Let me whip up a profiled build for you quickly here so you can see what i mean.

deadl0ck, 2019-08-01:
What sort of speed does the PocketGo / BitBoy run at normally ? And how damaging is the overclocking ? i.e. is it an issue to just go ahead an overclock and use your build

gameblabla, 2019-08-01:
> What sort of speed does the PocketGo / BitBoy run at normally ? And how damaging is the overclocking ? i.e. is it an issue to just go ahead an overclock and use your build

"Normally", the stock speed is 533 Mhz. As it is right now, most emulators/games are running at 702 Mhz right now, so they are already quite overclocking pretty far considering our pocket.
Now, i know that most PocketGo should be able to support 798 as well but it's still hit and miss and i just can't set the default config to 798 Mhz and call it done.

I have some commits i want to publish as i've made some minor optimizations before i give it a try again.

deadl0ck, 2019-08-01:
Cool - thanks for all the info and good luck with it ! It's about 20 years since I worked with C so I don't think I'd be much help. All Java and Python now.....

gameblabla, 2019-08-12:
[ngpcemu_bittboy_v1.zip](https://github.com/TriForceX/NewBittboyCFW/files/3493913/ngpcemu_bittboy_v1.zip)

Well my bittboy v1 broke so i'm releasing it with the executable being PGO'd.
Performance isn't going to be optimal but oh well, you were warned.

deadl0ck, 2019-08-13:
Great stuff @gameblabla - I'll hopefully get to it this week (if not, then then weekend).

deadl0ck, 2019-08-18:
No joy - I installed it and just ran it from the explorer app - it executed some command line stuff and then did nothing.
I'll post a video if it helps

gameblabla, 2019-08-18:
It doesn't have a menu so it also needs a command line argument, aka a rom, for it to launch.

deadl0ck, 2019-08-18:
D'Oh - Sorry - will give it a whirl. I'm on the latest CWF so I need to figure out how to add the app + rom as a command

deadl0ck, 2019-08-18:
Still no joy - I turned on the Logs and in the Log Viewer I get:

"Starting NGPCEmu
Segmentation fault"

gameblabla, 2019-08-18:
Well i'll get my PocketGo tomorrow, looks like a crash. Odd because i managed to make it work on my Bittboy v1.

deadl0ck, 2019-08-19:
Let  me know when there is a new version and I'll check it on my PocketGo also

deadl0ck, 2019-08-23:
Hey @gameblabla - were you able to recreate the crash on PocketGo ?

gameblabla, 2019-08-23:
Well this one is going to take longer than expected... It used to work but not anymore. I suspect it is caused by GCC 9.1 so i'll have to upgrade to GCC 9.2.

deadl0ck, 2019-08-23:
Thanks for the update !

gameblabla, 2019-08-30:
[ngpcemu.zip](https://github.com/TriForceX/MiyooCFW/files/3561129/ngpcemu.zip)
It's working now (it was really caused by GCC 9.1, GCC 9.2 fixed this) but it's still way too slow.
Maybe tweaking the flags will improve this a bit but i doubt.

deadl0ck, 2019-08-31:
Thanks @gameblabla - I'll take a look as soon as I get a chance

whiskyfly, 2020-11-16:
Hi, @gameblabla , work here in my pocket go. If you need help, please ask. Thank you.

gameblabla, 2020-11-16:
I know it's working and i had made many other changes after that but it would still require some other work, like switching to MAME's z80 core, for it to run faster.

whiskyfly, 2020-12-10:
> I know it's working and i had made many other changes after that but it would still require some other work, like switching to MAME's z80 core, for it to run faster.

Yes, if you need some help for switching z80 core, i can help.

gameblabla, 2020-12-10:
> > I know it's working and i had made many other changes after that but it would still require some other work, like switching to MAME's z80 core, for it to run faster.
> 
> Yes, if you need some help for switching z80 core, i can help.

Oh ? You can help with that.
As i said earlier, i wanted to switch to MAME's z80 core, more specifically this one from my SMS Plus GX fork
https://github.com/gameblabla/sms_sdl/blob/master/source/cpu_cores/z80/z80.c 
https://github.com/gameblabla/sms_sdl/tree/master/source/cpu_cores/z80

Right now NGPCEMU uses the z80 fuse core, which is ultra slow
https://github.com/gameblabla/NGPCemu/tree/master/mednafen/hw_cpu/z80-fuse

Anyway, if you think that's something you can do then be my guest !
Another alternative would be to have cz80 as an optional z80 core instead. This one is much faster, however the older version is not GPL. (there's a newer one under the GPL though)