TriForceX/MiyooCFW #11: Reduce firmware image size

ngsoftwaredev, 2019-04-12:
I would like to use a 4GB micro SD card, and currently the image requires about 96MB more than a 4GB size. I was able to  image my 8GB card and then trim the _main_ partition to flash it.

It would suffice that you lower the size of the _main_ partition from 3GB, to say 2 GB or even less. This would also make the download lighter ;-) 

And you'd still be able to flash it to larger cards and size it up after flashing.

amwmedia, 2019-04-12:
how much free space is left for roms after you reduce the main partition size? I think the most convenient approach for consumers would likely be to offer 3 different downloads (4GB, 8GB, 16GB). Though I'm not sure how difficult that would be.

ngsoftwaredev, 2019-04-12:
How much space is left for ROMs is up to your sd card size. It's easier to
have an image that fits on a minimum capacity, then flash it and extend the
partition afterwards to reclaim the free space. The resizing might even be
performed automatically on the first boot, as RetroFW for the RS-97 does.
It's fairly simple to do on your own though.

amwmedia, 2019-04-12:
yeah, if it can be done on first boot, that would be great. my comment about wanting different sized downloads was more for people who are not familiar with the idea of partitions, nor the ways to resize them. They just go through the steps to flash their card and wonder why it now has less space.

TriForceX, 2019-06-13:
The image size is not the problem, you need to extend the main partition. Once the image is burned on the SD card, open Partition Wizard and choose Disk Management. Then right click on the main partition and choose extend (Set it to the maximum) and apply.

closed 2019-06-13 by TriForceX