TriForceX/MiyooCFW #12: Cave Story game doesn't save anymore

francescoBreve, 2019-04-13:
I've been playing some Cave Story and the experience was quite smooth. After the boss fight with Enraged Toroko, the game directs me in a new area where i get a save point.  Here I try to save the game as usual but i notice the game doen't override the savefiles, nor it creates new one in the new slots. It says "save complete" but when you go check the save files the doen't change nor add to the list.

TriForceX, 2019-04-14:
This game port is under development, some things may not work.
We will keep you informed if there is any progress with it.

nicosensei, 2019-04-14:
Thanks for this update. I can't find a way to hide the mouse pointer in Cave Story, that's a bit annoying given the small screen size ;-p

francescoBreve, 2019-04-14:
Update, after turning the bittboy on and off the game now saves again, not sure what causes the issue. I also have the mouse cursor problem, i think it will be fixed in future updates. 
Keep up the good work!

closed 2019-06-13 by TriForceX