TriForceX/MiyooCFW #14: Screen tearing

dilbirin, 2019-04-14:
Hi again.I have a problem about 60fps on all emulators. All emulators have tearing on the screen. Is there a solution to this?

Thanks again for custom firmware.

dilbirin, 2019-04-16:
I found a hardmod on this site:

https://steward-fu.github.io/website/handheld/miyoo/fix_te.htm

But I guess, that's for v1.  I can't find on my bittboy v2.

TriForceX, 2019-04-16:
Yeah, this is a hardware problem, it wont be fixed until the 3rd release of the console (rev3).
We are working to make the guide for rev1, i dont know if theres one for rev2.
Stay tuned.

alexandershalin, 2019-04-26:
Is it correct that this issue was not fixed in rev3?

krasny2k5, 2019-05-06:
I also would like to know if v3 fixes the screen tearing problem.

TriForceX, 2019-05-06:
@alexandershalin @krasny2k5 Bittboy developers decided to delay the old v3 and they already released a different v3 (ghost key fix + glass screen), but screen tearing was not fixed yet...

closed 2019-05-06 by TriForceX

chipxsd, 2019-06-04:
Can you give more detail on the hardware issue? What's the TE pin? Is it the thing that selects the 30Hz/60Hz refresh rate?

I wonder how the screen is driven anyway? I'd love to see the schematics, if they're available?

TriForceX, 2019-06-04:
> Can you give more detail on the hardware issue? What's the TE pin? Is it the thing that selects the 30Hz/60Hz refresh rate?
> 
> I wonder how the screen is driven anyway, I'd love to see the schematics, if they're available?

You can get more info at #new-bittboy channel on our Discord https://discord.me/retrogamehandhelds Some developers like @steward can give you more details about it, but i warn you no one is interested in the Bittboy development anymore due the main company doesn't support us...