TriForceX/MiyooCFW #16: Issue with assigning a drive letter to main after flashing card

EnlistedGI, 2019-04-22:
[image]
I have followed every tutorial to get my 32GB card setup correctly, but still have the same issue. I do not get a drive letter attached to the ‘main section’ after flashing the card and when I try to using minitool partition it does not apply it. I have used windisk imager and etcher both give me the same results. So my card only reads a few hundred mb not the 28GB I have as an expanded main section.

TriForceX, 2019-06-05:
Did you removed all partitions and format in FAT32 before write the image? i recommend you to use SD Card Formatter

mougino, 2019-06-10:
Exact same issue, using Win7 and Bittboy V2.5, deleted all volumes and formatted the whole sd (32 GB) as a prerequisite. After flashing the V3.8 beta image, only the "Boot" partition (256 MB) can be assigned a letter. Assigning a drive letter to the "Main" partition doesn't work (Win admin tools, Partition Magic). Even stranger: "Boot" volume already contains 4 subfolders gb, gbc, gba and nes right after flash, but once filled with ROMs these are not recognized by the Bittboy.

TriForceX, 2019-06-13:
You may not touch the boot partition, you need to place stuff on main partition. 
About your SD card you may remove all partitions and format with something like SD Card Formatter or Partition Wizard. Next once the image is burned on the SD card using win32disk imager, open Partition Wizard tool and choose Disk Management. Then right click on the main partition and choose extend (Set it to the maximum) and apply.

closed 2019-06-13 by TriForceX