TriForceX/MiyooCFW #20: not working with CFW 3.6

PredatorMP, 2019-05-08:
Just got my new BittBoy. Strange.......it is not working with CFW 3.6. No display (black screen)

TriForceX, 2019-05-08:
Whats your console revision? v2 or v3? we have two different guides for each one, also for v2 we recently fixed the cfw 3.6 link.

PredatorMP, 2019-05-08:
V2

TriForceX, 2019-05-08:
Can you try to install the recently updated we did? https://github.com/TriForceX/NewBittboyCFW/blob/master/Rev2-Guide-EN.md

PredatorMP, 2019-05-08:
I try this but i have black screen :/

TriForceX, 2019-05-08:
have you formatted the SD card with win32diskimager tool? remember need to be in Fat32

PredatorMP, 2019-05-08:
yep

TriForceX, 2019-05-08:
do you hear the boot-logo sound? and another question, the screen turns on? (black but with the backlight on)

PredatorMP, 2019-05-08:
yes i hear the sound

TriForceX, 2019-05-08:
if it's black screen but you hear the boot-logo ding, it's an issue with the graphics driver
which would be either the wrong image, or something wrong with the image or sd card
I'd verify the file they flashed, and if correct flash again

PredatorMP, 2019-05-08:
i solve the problem

closed 2019-05-08 by PredatorMP