TriForceX/MiyooCFW #24: ‘Select Scaler’ crashes gambatte_sdl

timothysob, 2019-05-12:
Choosing ‘Select Scaler’ in the GBC emulator crashes gambatte_sdl MOST of the time. Not every time, oddly enough. It seems if I interact with other menu options first, it may not crash.

TriForceX, 2019-05-14:
Which CFW version are you using? also which Bittboy version?

timothysob, 2019-05-14:
I’m using a BittBoy V3, firmware 3.6.

TriForceX, 2019-05-14:
If you can try the _CFW 3.7 Beta 2_ would be nice because a new **gambatte** was released there solving some problems.

timothysob, 2019-05-14:
Will do! I’ll let you know.

timothysob, 2019-05-17:
OK, I updated to 3.7 beta 2. Choosing 'select scaler' still crashes the GBC emulator, and I just discovered that it also crashes the new 'rumble' gb emulator.

TriForceX, 2019-05-17:
is the menu text is uppercase or lowercase?

TriForceX, 2019-05-17:
is the menu text is uppercase or lowercase? the steward one has lowercase text, while the hi-ban has uppercase text, the last updated was from hi-ban.

timothysob, 2019-05-17:
OK so the problem seems to be solved in the version with ALL CAPS menu items (If I go to CREDITS, it says BUILD 20190512-215230), I can choose whatever scaler I want with no issue.

I used to have two or three different GB emulators that would show up for me -- I believe they were called Gambatte, Gambatta-SDL, Gameboy-Rumble. I had already deleted the other duplicate GB emulators that were crashing, so I can't test those other versions now to see if they'd still crash. I'm fully working with this version though! It seems to not have the same problem.

TriForceX, 2019-05-17:
Nice to hear u got working now. we know that having those versions around there is very messy, it should be only one, but is hard to deal with the Bittboy people to get this a little bit organized.

The gambatte by hi-ban is the last updated one.

closed 2019-05-17 by TriForceX