TriForceX/MiyooCFW #25: ZX Spectrum / FUSE

cmgonzalez, 2019-05-24:
Is posible to add FUSE to the CFW? I saw running on the LDK console.

TriForceX, 2019-06-05:
Yes is possible, the emulator available is Speccy and the source code is here:
https://github.com/pingflood/speccy

So it can be ported to Bittboy 😄

Andrew-H2O, 2019-07-04:
+1 for this - would appreciate a speccy emulator if someone would see fit to add it in.

roodyster, 2019-09-11:
I would also very much like to see https://github.com/pingflood/speccy ported to CFW

meowfiz, 2020-04-02:
Are there any chances for porting Speccy emulator for PocketGo v1?

TriForceX, 2020-04-02:
> Are there any chances for porting Speccy emulator for PocketGo v1?

yes, would probably run well on the hardware, thats why i didnt closed this request... but someone needs to do the port.

cmgonzalez, 2020-04-03:
Hi do you know of some doc or tutorial to port, I can help on this...soy chileno :)

TriForceX, 2020-04-03:
> Hi do you know of some doc or tutorial to port, I can help on this...soy chileno :)

Read here https://github.com/TriForceX/MiyooCFW/wiki/Making-Games
Also you can join our Discord at https://discord.gznetwork.com to talk with other developers who already ported stuff.

PS: Im Chilean too :)

cmgonzalez, 2020-09-22:
This emulator works perfect!
http://lessermatters.homeunix.com/PocketGo/

salvacam, 2020-10-05:
New release with fix full screen and mapping joystick

meowfiz2020, 2020-10-05:
New update is an awesome idea (custom mappings & full screen) but there is an issue with open file menu. When I select it with a "B" button a file list shows up, but can't select a file - B button closes a window without selecting a file, A button moves selection one row up.

salvacam, 2020-10-05:
> New update is an awesome idea (custom mappings & full screen) but there is an issue with open file menu. When I select it with a "B" button a file list shows up, but can't select a file - B button closes a window without selecting a file, A button moves selection one row up.

You must have the option "AUTO PLAY IMAGE" with ON

meowfiz2020, 2020-10-05:
Thanks! That works like a charm 👍 
The only thing that might be a problem in some games (Galaxian) is that you can't map CapsShift to joystick button.

salvacam, 2020-11-03:
> Thanks! That works like a charm 
> The only thing that might be a problem in some games (Galaxian) is that you can't map CapsShift to joystick button.
New rela
https://github.com/TriForceX/MiyooCFW/issues/25#issuecomment-703662956

meowfiz2020, 2020-11-05:
Tested. Works great!

meowfiz2020, 2020-11-09:
I'm still having some issues. When I try to modify custom joystick settings game freezes and I can't do anything. I can popup menu and hoover over emulators  options but only choice I can make is to Quit emulator. (Using on PocketGo)

salvacam, 2020-11-09:
> I'm still having some issues. When I try to modify custom joystick settings game freezes and I can't do anything. I can popup menu and hoover over emulators options but only choice I can make is to Quit emulator. (Using on PocketGo)

Are you using the latest version? https://github.com/salvacam/speccy/releases/tag/0.1.8

meowfiz2020, 2020-11-10:
Yes it's 0.1.8. I've checked both normal/full screen versions (419912/41908 bytes):
The effect:
https://youtu.be/fD37XGAWHuw

salvacam, 2020-11-10:
> Yes it's 0.1.8. I've checked both normal/full screen versions (419912/41908 bytes):
> The effect:
> https://youtu.be/fD37XGAWHuw
Make sure to download the latest release.
Should be seen in the menu "Pause (R2 L+B)" y "FULLSCREEN (L2 L+A)"

meowfiz2020, 2020-11-13:
Ok. The problem is fixed. It was quite bizarre, because I had  several versions of emulator binary in speccy directory on pocketboy. I had to delete all binary files except "pocketSpeccy" to force pocketboy to select the proper one. Thanks a lot - no more freezing!

salvacam, 2020-11-28:
> Is posible to add FUSE to the CFW? I saw running on the LDK console.

There is already a version of the Fuse emulator for Miyoo devices, please note that it is a preview version and may be buggy.
https://github.com/salvacam/fuse-od/releases/tag/0.0.1Miyoo

salvacam, 2021-03-06:
This issues can be closed.

closed 2021-03-07 by TriForceX