TriForceX/MiyooCFW #26: L2 R2 can‘t  work

sunxuold, 2019-05-29:
i use latest 3.8 version L2  and R2 can't work  they use same key map with B and TB
my bittboy is   V2  version. 
i replace zImage with another version L2 and R2 can work but speaker can not be turn off
when i use earphone.
can i get one zImage file support L2 R2 without earphone issue?

TriForceX, 2019-06-05:
Did you follow the pictures on the wiki? https://github.com/TriForceX/NewBittboyCFW/wiki/Shoulder-Buttons

sunxuold, 2019-06-06:
yes we are use same way for add LR12.   and ii think my bittboy is v2.5 because the two position no any chip ，the PCB  is empty。

sunxuold, 2019-06-06:
and for my know  Steward use the key for ghost key issue may it cause the problem.

TriForceX, 2019-06-13:
Can you send a picture from your bittboy PCB?

sunxuold, 2019-06-14:
sure, below is my bittboy's PCB.

sunxuold, 2019-06-14:
Below is my PCB

[image]

sunxuold, 2019-10-07:
> Can you send a picture from your bittboy PCB?

any solution can make L2 R2 work?

Apaczer, 2023-12-25:
The earphone issue has been solved a few commits ago. You migth give it ago for latest BETA version, the L2/R2 mapping also should be valid (not sure cuz of your custom setup).

closed 2023-12-25 by Apaczer