TriForceX/MiyooCFW #27: is there any MSX EMULATOR...

funkyman0, 2019-06-06:
I searched digux emul... but any can adopt bittboy...
probably I listend there is MSX Emul... somewhere...
am i confused Thing...
(-_-)a

TriForceX, 2019-06-06:
The emulator available is Dingoo's fMSX and the source code is here:
https://github.com/pingflood/dingux-msx

So it can be ported to Bittboy 😄

funkyman0, 2019-06-06:
Thanks for  your concern...
I'll try it.
Thanks again.

roo1234, 2019-09-14:
there's also 
https://github.com/openMSX/

How can one compile this to miyoocfw?

gameblabla, 2019-09-15:
Only an older version of OpenMSX is doable as they removed any kind of support for SDL 1.2.
I think Dingux-msx would be a safer bet. I can look into it.

TriForceX, 2021-03-07:
Theres Bittboy MSX now: https://github.com/pthalin/bittboy-msx/releases/latest

closed 2021-03-07 by TriForceX