TriForceX/MiyooCFW #32: SPI image not available

mmaret, 2019-06-14:
In the Rev1 guide (https://github.com/TriForceX/NewBittboyCFW/blob/master/Rev1-Guide-EN.md), the spi image required is not available anymore (https://github.com/steward-fu/miyoo_rel)

TriForceX, 2019-06-17:
I will upload today, thanks!

TriForceX, 2019-06-20:
hi, i talked with steward and hes rearranging his repositories to get all back. i will post here soon

TriForceX, 2019-06-24:
due the delay i found another repo with the code, but i warn you is an older code (not the last one by steward)

https://github.com/johnaa123/miyoo_src
https://github.com/johnaa123/miyoo_dev
https://github.com/johnaa123/miyoo_rel

closed 2019-07-17 by TriForceX

gameblabla, 2019-08-13:
It's available here : 
https://github.com/ospqqcom/miyoo_rel/raw/master/miyoo_stock_v1.bin

DomoKoder, 2019-08-13:
And how I flash it back? The same way I flashed the custom? 

sudo sunxi-fel -p spiflash-write 0 SPI image .bin
ex> sudo ./sunxi-fel -p spiflash-write 0 miyoo_stock_v1.bin

Thanks in advance :)

gameblabla, 2019-08-13:
Yes, just like that. Make sure its in FEL mode too.

DomoKoder, 2019-08-13:
How can I ensure that? Isn't enough the first command with sudo?

gameblabla, 2019-08-13:
FEL mode is basically when you set the bittboy to be set in a mode where it can be flashed.
Basically the jumper wire on Ground and Pin 2. Once you boot it up with said wire, it will appear as a USB device.
You should know, since you flashed the CFW Spi firmware image to your bittboy...

DomoKoder, 2019-08-13:
Sure, got it no prob.. Just didn't know it's called FEL mode :)