TriForceX/MiyooCFW #33: Is there any guide for compiling CFW and emulators?

forbbs, 2019-06-22:
Do not know how to compiling it.

TriForceX, 2019-06-24:
You need the buildroot to compile it (is already linked on the main repo). 
Feel free to join our chat to get some support https://discord.me/retrogamehandhelds

TriForceX, 2019-06-24:
You need the buildroot to compile it (is already linked on the main repo). the resulting toolchain will end up in output/host.

Feel free to join our chat to get some support https://discord.me/retrogamehandhelds

forbbs, 2019-06-26:
yes.I have build toolchain.
Is it lacks .config and device-tree for kernel compiling?

closed 2019-08-08 by forbbs

TriForceX, 2019-08-08:
We recently added two guides, it may be usefu for you.
https://github.com/TriForceX/NewBittboyCFW/wiki/Build-Source
https://github.com/TriForceX/NewBittboyCFW/wiki/Making-Games