TriForceX/MiyooCFW #35: PICO-8 and LOVE2D

italomaia, 2019-06-28:
Hi. How can I compile PICO-8 and LOVE2D for bittboy?

TriForceX, 2019-07-11:
Theres no guides for that yet.

But mainly you need the buildroot to compile it (is already linked on the main repo). The resulting toolchain will end up in output/host.

Feel free to join our chat to get some support https://discord.me/retrogamehandhelds

gameblabla, 2019-07-24:
Pico-8 is impossible, trust me. The whole thing is proprietary.
The only open source alternative for it, picolove, requires OpenGLES 3D acceleration.
I had made a port for the GCW0 and even there it requires updated drivers for it to potentially work.

If you really want to play pico-8 games on your bittboy, then you'll have to ask for the source code to lexaloffle... Good luck with that.

closed 2019-07-24 by gameblabla

gameblabla, 2020-09-20:
I ported Retro-8, a pico8 interpreter for the bittboy/LDK/GKD350H :
https://gameblabla.nl/files/ipk/bittboy/retro8_bittboy.zip

Compatibility isn't perfect but will allow you to play some games.
I think we can remove the "wontfix" and close it for good.

cmgonzalez, 2020-09-23:
I've tried it on a bittboy (ver3) 

When run from gmenu2x by picking a p8 file seems to run, but with black screen, I can only hear music from the game.

Launching from explorer retro8_bittboy.elf closes explorer.

gameblabla, 2020-09-23:
> I've tried it on a bittboy (ver3)
> 
> When run from gmenu2x by picking a p8 file seems to run, but with a black screen, I can only hear music from the game.

Depends on what game you were trying to run.
https://docs.google.com/spreadsheets/d/1pSrwKAXKLSApBIcbpgIgwuPSfiHzM1dyDt_m8bPnQG4/edit?usp=drivesdk

Not all games will work sadly but popular ones like Celeste will though. There's also Taco8 that i ported but this one doesn't support p8.png files and has no sound 
[tac08_bittboy_gameblabla.zip](https://github.com/TriForceX/MiyooCFW/files/5271605/tac08_bittboy_gameblabla.zip)

> Launching "retro8_bittboy.elf" from explorer, force close of explorer.
Yup that's normal : it needs a game to run.

Tenchirox, 2021-06-24:
Newers tac08 revisions seem to handle png any posible update ?

gameblabla, 2021-06-24:
[retro8_bittboy_2021_revision.zip](https://github.com/TriForceX/MiyooCFW/files/6712542/retro8_bittboy_2021_revision.zip)

Here's one with the newer changes plus a fix from me fixing Hungry Harry 3D. (though the sprites are not flipped but it works fine otherwise)

gameblabla, 2021-06-24:
> Newers tac08 revisions seem to handle png any posible update ?

And the newer tac08 revision are not open source. So how do you expect anyone to port tac08 without any source code? The last changes date back to March 2020, before the new PNG support or even sound support.

blurymind, 2021-07-14:
how come it works on arkos and emuelec then?  Did the pico8 dev port them himself?

gameblabla, 2021-07-16:
> how come it works on arkos and emuelec then? Did the pico8 dev port them himself?
> 
> Why did you close this issue?

It "works" on there because they can run the official binary with no issue. We cannot do that because the official binary was built for ARMv7 + hardfloat, and our platform is ARMv5. The only way (for now) to play some of the Pico-8 games is to use my Retro8 player, which is already part of the latest CFW release.

szymor, 2023-05-01:
Isn't FAKE-08 mentioned in README.md of this repo?

szymor, 2023-05-01:
If you feel to do so, then yeah, sure. Though it can be read from the official github repo of fake08 as well.