TriForceX/MiyooCFW #38: image too big, no "main" partition

yepyepyep4711, 2019-07-02:
Hello,

I've just gotten my bittboy 3.5. I wanted to update to the latest software following [this](https://github.com/TriForceX/NewBittboyCFW/blob/master/Update-CFW-EN.md), but there was no "main" partition on the sd that was delivered. So I tried the installing procedure.
So I deleted all partitions on the SD card that was included, then formatted to FAT32.
Sadly, Disk Imager tells me the image I [downloaded](https://github.com/TriForceX/NewBittboyCFW/blob/master/Rev3-Guide-EN.md) is too big. Not by much (90112 sectors, 88KB), but still. It also tells me there seems to be data on the extra space.
I tried anyway, and after the update I face the same issue as before, only the boot partition is visible to windows, the other three are only visible to the partition manager. Neither MiniTool Partition Wizard nor the windows disk manager nor EaseUS Partition Manager were able to give the partition a drive letter, it always reverts back to no letter and hence I can't write to it.
I went to the trouble and mounted the .img as a HDD and resized the last partition, the "main" one. Sadly, I don't know how to resize the .img itself, and I got the same warning. The new .img shows empty space after the "main" partition, but the problem stays. No letter, no file access.

I've tried with the sandisk that was delivered as well as with another 8GB I had lying around.

At this point I'm up for ideas.

- can someone make a smaller .img?
- does someone have a link to the previous version so I can try that?
- any other ideas? (not "try a bigger sd", please, I'm doing that already but it takes ages to back it up first because it's a big one; I'll report when it's done)

yepyepyep4711, 2019-07-02:
ok, there's a problem with the .img
Writing to the bigger SD obviously doesn't produce a size error, but still no "main" partition.
Please note I have now tried this with three different cards and two different card readers

TriForceX, 2019-07-04:
what are you using to format sd card?

yepyepyep4711, 2019-07-04:
whatever I'm using for doing the partition. But it shouldn't play a role, right? It's only important to have a partition before writing the image in order to have a letter to address in Disk Manager, no? That partition is going to be erased ba writing the image anyway.

TriForceX, 2019-07-04:
it should, also it depends your operative system too, etc...

which OS are you using?

i think you're confusing how the things works, thats why i need to know all the scene.

yepyepyep4711, 2019-07-04:
it's win7 64 bit.

Where am I going wrong?

yepyepyep4711, 2019-07-04:
I went the other way and made an .img from the SD card I had written. It's a bit under 46 MB smaller.

TriForceX, 2019-07-04:
When you install the CFW for the first time, you need to expand the "main" partition, because after .img is written, the SD card got divided in 4 partitions: 

- unlabeled1 (256mb)
- unlabeled2 (256mb)
- boot (256mb)
- main (around 3 GB)

So, if theres more free space you will not see, because is unallocated, you need to "extend" the main partition to increase its current size. I think thats your problem, some stock sd cards doesnt comes with the expanded space, so you have to do it manually.

If you will do this from scratch, is required to not have more partitions on your SD card, also i dont recommend you to use the windows disk management because in some win 7 not all partitions appears by default (or are unlabeled).  I recommend you to use SD Card Formatter to remove all partitions and format to FAT32 properly, and Partition Wizard to extend the main partition after .img write.

glebtv, 2019-07-04:
@yepyepyep4711 i think you need windows 10 or linux to see main partition. See: https://www.google.com/amp/s/amp.reddit.com/r/Bittboy/comments/c8aqhj/add_games_on_pocket_go_in_a_windows_7_computer/

yepyepyep4711, 2019-07-04:
@glebtv wow, that would suck. @TriForceX, do you know if this is true? I've looked and others are having the same problem. I don't get it windows is able to show several partitions on a drive. What is different here?

TriForceX, 2019-07-04:
@yepyepyep4711 im not sure (cant test right now). can you see the main partition on windows disk management? if you see it there can you assign a letter? it will not affect anything

yepyepyep4711, 2019-07-04:
I can see it alright in a partition manager. But no luck assigning a letter. Depending on the tool I use, either I get an error message (windows disk management) or the the letter doesn't stick EaseUS Partition Manager

gameblabla, 2019-07-04:
Yup, before Windows 10 Creator's Update, it would not properly support sd cards with multiple partitions on them and only the first one would get mounted.
On Windows 7, you can use RMPrepusb to change the order of a partition and change it back afterwards but it's more of a pain than anything and i would recommend you just upgrade to Windows 10 or use any Linux distributions.

You can thank Micro$oft for that, this was intentional on their part before they decided to change their mind.

yepyepyep4711, 2019-07-05:
@gameblabla hey, thanks for the RMPrepusb tip! That works. You know what I've been doing so far if I wanted to change something? Make an image of the drive, mount it in writable mode, then write the image back. 8GB to move to and fro, plus extra steps for mounting, un-mounting and such. This is way faster.

yepyepyep4711, 2019-07-05:
so if this can't be fixed we're nearly done with this issue. The last thing left would be my recommendation to cut the free space on the fourth partition by 50MB so that people don't get a warning about size, as it seems actual size is dependent on the SD card manufacturer.

TriForceX, 2019-07-16:
Theres another alternative to access the "main" partition on Windows 7 or 8, please read here:
[image]

closed 2019-07-16 by TriForceX

yepyepyep4711, 2019-07-16:
yup, that works basically the same as RMPrepusb