TriForceX/MiyooCFW #39: [BUG] CPU Clock speed settings per single emulator/app/program

generalerusso, 2019-07-03:
Hello, just recently managed to get my hands of the PocketGo and with the most recent CFW I noticed that the CPU speed per specific emulator can be changed from the stock 702. Trying to test it on (for example) the SNES emulator to see if some betterment in FPS/speed could be seen, i put it at 750.

Once i then select a game, the entire system seems to lock up, no buttons responds, which makes me need to do an hard shutdown via the on/off slider. Once i get back in, and put the CPU settings to stock (by pressing select) the game starts without any problems.

TriForceX, 2019-07-03:
Thats not a bug, more CPU frequency doesnt mean it will run better, it depends how the emulator/app runs and how much speed can handle. You can change "all emulator/app" cpu speed in main settings or "per emulator/app" settings using the select button, so it will not affect the others.

Is a normal fact that sometimes the emulator/app will crash or not load, because this is an "trial and error" case to figure out which cpu speed will run better or not.

closed 2019-07-03 by TriForceX

generalerusso, 2019-07-03:
Thank you for the answer, I will do more testing!