TriForceX/MiyooCFW #40: cleaning up the interface

yepyepyep4711, 2019-07-03:
Hi,

I bought this thing for one system only: Game Boy (and GBC but that's the same emulator).
Also, I want my kids to be able to use it, so less is more. Is there an easy way to get rid of all sections but emulators and get rid of all emulators except the few I want?

Many thanks

TriForceX, 2019-07-03:
Yea of course, we was discussing it today in our chat, because the current cfw package has a lot of stuff and is too huge. i have not been generating the previous cfw packages, but personally i dont like how it has been doing.

So i will make a cleaned image with the emulators/app/skins separated.

yepyepyep4711, 2019-07-04:
Great! If I'm very lucky it will solve [my other problems](https://github.com/TriForceX/NewBittboyCFW/issues/38) ;)

yepyepyep4711, 2019-07-05:
any way to have the bittboy logo stay on during the whole boot, or generally avoid seeing the matrix-like text output each time you do pretty much anything?

TriForceX, 2019-07-05:
yea maybe, but can you post this as another request? so we can add it as another enhancement. thanks.