TriForceX/MiyooCFW #41: background noise

yepyepyep4711, 2019-07-05:
hey all,

I don't know about you, but my Bittboy v3.5 gives off rather high-pitched noise pretty much as soon as the starting logo gets refreshed (it blinks, you'll know it when you see it).
Now I know this sounds like a hardware problem, but I as I said, it's not straight from the start. Also, it seems to vary, but I don't know with what. Charge maybe? 
At some point I noticed it would disappear when the green "bitt boy" screen appeared, but not at the moment. Now it stays on even while playing :( Any ideas?

TriForceX, 2019-07-05:
Yes, is a hardware issue and is directly related with the screen, try to change the brightness to less or high, you can hear how the noise increase/decrease

yepyepyep4711, 2019-07-05:
ah, that was that. So many thanks! Incredibly enough, the best setting is at full brightness. Bad for the battery, but otherwise not the worse that could happen...

TriForceX, 2019-07-05:
Ok, the reply i posted was from the official reply about it. But i recently found another way to partially fix the problem. 

This will change the frequency setting of the PWM driver. Basically, it will switch the backlight LED on and off at a much higher rate, which should move a portion of the audio noise up into an inaudible frequency. As far as we know, this shouldn't be able to cause any harm to the hardware, but no one will accept any responsibility for any damage that may occur. 

Use this at your own risk! Link here: https://byteporter.com/pocket-go-pwm-script-install/

yepyepyep4711, 2019-07-05:
wow, that *is* very interesting. Do we have anyone from the hardware side that could say whether or not this could be harmful in the long run?

yepyepyep4711, 2019-07-10:
just re-opening this for the following question. I realize my hardware question has been answered already, but in my simple mind, upping the frequency would also increase the power consumption, correct? So there would be no difference between upping the frequency and having the display at full power all the time with the regular setting? Or is playing at full brightness still more power hungry than this?

yepyepyep4711, 2019-07-24:
> Ok, the reply i posted was from the official reply about it. But i recently found another way to partially fix the problem.
>
> This will change the frequency setting of the PWM driver. Basically, it [...]

I went and did this. It works (except for the part where renaming a link is a bit harder on a bittboy than on a pocketgo, see https://github.com/TriForceX/NewBittboyCFW/issues/44, but hey). I'm blown away! I can turn down the light and there's no noise! Fantastic! Thx to all involved!

If you try it yourself, don't get scared if it takes a few "black restarts" as in https://github.com/TriForceX/NewBittboyCFW/issues/52. At least it happened to me, but it worked in the end.

I'll try and monitor power consumption.

yepyepyep4711, 2019-09-05:
I have nothing bad to report on this mod. Just works.

yepyepyep4711, 2020-03-21:
hello everyone. Just reopening this so that at least some people notice. I have now run this at 17kHz all these months until yesterday. Yesterday, suddenly, the screen went dark. Cycling the device showed consistent behaviour: the device would start, the screen would activate but go dark at the moment the "Bittboy" logo would normally scroll over the green background. I will have to assume this is the moment the frequency setting kicks in. Here is how I know:

Luckily, I have another Bittboy, so I put the SD card in it and turned it on. It started fine, so I went and reverted the frequency settings to the original value, and put the card back in the device that had the problem. It started and played fine. I went back to 17kHz, and the screen went black again.

Long story short: this modification might very well hurt your Bittboy and/or stop working. Unless we known long-term working configurations (frequencies and corresponding hardware revisions), I would advise against it.

Cheers

symbios24, 2020-07-31:
The real gameboy had this problem also :p

akg83, 2023-05-31:
@TriForceX if this is a hardware issue, why is this issue still opened? should be moved to the discussion area? there all possible solutions of "use at own risk" can be discussed, just a suggestion to clean up the issues list ;-)

TriForceX, 2023-05-31:
@akg83 it was still opened because i cannot remember all issues posted by people 😄 

also, if an hardware issue can be "fixed" through code/scripts is legit to have them here, but yea, the workaround was already posted, i just forgot to close it.

Reference: https://github.com/TriForceX/MiyooCFW/wiki/Sound-Noise-Fix
thanks for the reminder!

closed 2023-05-31 by TriForceX