TriForceX/MiyooCFW #42: do game saves work?

yepyepyep4711, 2019-07-05:
Hello,

always on the lookout to make this thing as user friendly as possible, I've been looking at game saves vs saved states in the CFW's gambatte. So far, in-game saves seem to work fine (example: Super Mario Bros Deluxe, Pokemon Yellow), so that saved states would only be needed in games that don't save, like high-score keeping in tetris.

Is this correct or am I missing something? In-game saves are so much better than saved states for the kids.

TriForceX, 2019-07-05:
are you shutting down properly the console? (exiting the emulator and power off from main settings)

yepyepyep4711, 2019-07-05:
Me, yes. I'm not sure the kids will always, though. Why?

yepyepyep4711, 2019-07-05:
truth be told, I sometimes do the following: exit the emulator properly and then just turn the device off. Or exit the emulator, press r to suspend, then turn off.

Why?

TriForceX, 2019-07-05:
Is wrong to "suspend" and next "turn off", you need "power off" from the settings options and then when the screen goes black you "switch off" the console.

This is not a console recommended for kids, remember this device runs a version of Linux, so thats basically a "mini computer", is not like a gameboy where you just on/off and plug game catridges... 

So, just "switch off" the console may currupt your SD card, is like unplug a computer from energy (or an USB stick) while is reading/writing data... something can be fucked up.

yepyepyep4711, 2019-07-05:
yes, I know the theory, but how does this relate to in-game saves? As long as I exited the emulator, the only thing that can be fucked up is the system, right? Not the saves?
My question was: can I rely on the in-game saves if I exit the emulator correctly (it looks like it so far at least in gambatte) or is it risky?

yepyepyep4711, 2019-07-06:
ok, I'm still interested in an answer, but I guess there is no way around using the saved states in my case, at least for some games. For example, Pokemon on Game Boy enables you to create a character and save its progress, but only one. And I have two kids, which means either a second Bittboy ;), or saved states.

gameblabla, 2019-07-24:
> can I rely on the in-game saves if I exit the emulator correctly (it looks like it so far at least in gambatte) or is it risky?

It doesn't matter if it's an ingame save or save state because an improper shutdown can corrupt any files that were still not flushed.

yepyepyep4711, 2019-07-24:
> > can I rely on the in-game saves if I exit the emulator correctly (it looks like it so far at least in gambatte) or is it risky?
> 
> It doesn't matter if it's an ingame save or save state because an improper shutdown can corrupt any files that were still not flushed.

I don't see a valid reason to keep anything related to a certain game in cache after the emulator running it has been exited. Do we know this to happen?

gameblabla, 2019-07-24:
> Do we know this to happen?

Because the filesystem does it async rather than in a synchronous way. This could be fixed by changing the fstab file and tell it to mount it using "sync" rather than "async".

yepyepyep4711, 2019-07-24:
> > Do we know this to happen?
> 
> Because the filesystem does it async rather than in a synchronous way. This could be fixed by changing the fstab file and tell it to mount it as "sync" rather than "async".

Are there any reason not too? Would that be an easy change?

closed 2021-10-15 by TriForceX