TriForceX/MiyooCFW #43: hide ugly command line screens

yepyepyep4711, 2019-07-05:
Hi,

as referenced [here](https://github.com/TriForceX/NewBittboyCFW/issues/40), this is a request to hide all the scrolling text appearing at boot or or when you load or quit a game.

At least during boot, my suggestion would be to hide it behind the "bitt boy" logo.

TriForceX, 2019-07-05:
Techincally the "bittboy logo is just a post-load animation that can be removed. Also  i was planning to replace the first boot logo (NXHope) with the Bittboy logo, because having two boot screens is too much, atleast for me.

kr1pt0ph0b14, 2019-07-05:
Having Bittboy animation as the only/single boot intro would truly be cool. 👍

yepyepyep4711, 2019-07-05:
yes please. Bittboy logo only, and no text walls at all. That would be ideal

yepyepyep4711, 2019-07-05:
let me correct myself. Since NES emulation is not an option on the Bittboy (see https://github.com/TriForceX/NewBittboyCFW/issues/45), the ideal thing would be to go straight from boot screen to game list, if possible a list that shows screenshots or covers of games so that it's pretty. That last part might be more work, but launching gambatte straight after boot is probably a matter of one line in a crontab, no?

yepyepyep4711, 2019-07-09:
> let me correct myself. Since NES emulation is not an option on the Bittboy (see #45), the ideal thing would be to go straight from boot screen to game list, if possible a list that shows screenshots or covers of games so that it's pretty. That last part might be more work, but launching gambatte straight after boot is probably a matter of one line in a crontab, no?
> 
> Then I would put all games (GB & GBC) in the same folder, and lo! Huge gain in user-friendliness. Boot, select game, play.
I've now come to realise what I'm looking for would be to start the selector after boot, not an emulator. I've achieved cover previews with this https://github.com/m1024x/retrogame-preview-guide
(I've used 128x128 for size). It's nice, but still not exactly "pretty". Maybe there are skins that work better with it. Happy for pointers.

So, once the "Bittboy logo everywhere" issue is fixed, what would be needed is a "start selector in rom folder automatically after boot"-option. I'll open a separate issue.

closed 2022-09-09 by Rezzy-dev