TriForceX/MiyooCFW #44: R for reboot is not working

yepyepyep4711, 2019-07-05:
Hello,

in the "shutdown properly" dialogue, there is a choice between Reboot (R), Cancel (B) or Poweroff (A).

(A) or (B) works, but (R) doesn't do anything.

TriForceX, 2019-07-06:
yes, it doesnt work, confirmed.

yepyepyep4711, 2019-07-24:
I know why it doesn't work. The "R" that is meant here is the "R" from the right shoulder button.
You have the same issue if you try and edit a link title. "L" is backspace, "R" is space. But the only "R" you have on a bittboy is not the shoulder button and goes in suspend. This is how I found out what "R" meant.

Apaczer, 2023-03-14:
yes "R" in gmenu2x stands for Right Shoulder R1 button and is defined by ``section_next=`` input keyboard value. You could change that value to other working or you could activate R1 button on bittboy with [hotkey combo](https://github.com/TriForceX/MiyooCFW/wiki/Miyoo-CFW#console-hotkeys).

closed 2023-03-14 by Apaczer