TriForceX/MiyooCFW #45: nes emulation choppy

yepyepyep4711, 2019-07-05:
Hello,

I've tried playing super mario bros with the included fceux, and the animation is very choppy.
Is that a hardware limitation or is there anything that can be done with settings?

TriForceX, 2019-07-05:
Is the screen-tearing problem, and yes, is a hardware issue, is the most known issue from the first Bittboy revision, the screen cant handle more than 30hz (is supposed to be 60hz), so the tearing occurs because it is trying to refresh twice.

TriForceX, 2019-07-05:
Its the screen-tearing, issue reported from the first Bittboy version, you can read more at #14

closed 2019-07-05 by TriForceX

yepyepyep4711, 2019-07-05:
thanks for the info. Any idea why this is not an issue for Game Boy games?

TriForceX, 2019-07-05:
Its depends of the emulator or game, in any can be different, for example for me is not very noticeable on neogeo or pc-engine...

About in gameboy specific the screentearing was reduced a little with some code tweaks, is less but still present. 

For some people is more noticeable than others...