TriForceX/MiyooCFW #46: Incorrect keymap

Konstantin22222, 2019-07-06:
Hi, recently i have bought pocket go and i camt normally use because in every program and emulator randomly keymap. And randomly menu button. Why not reset button been ALWAYS as exit??? Why in some emulators a is b and b is a? And av is not working, but this is not critically. If you fix keymaps the system will be nice.

krieg666, 2019-07-07:
I have a similar problem, with menu, keymap is wrong and can not to use with normally, can not to choose any emulator and use the console...

TriForceX, 2019-07-11:
Thats a common fact... in basic words, thats why the PocketGo is an "modified" version of the Bittboy, but the internal system is still the same... except they changed the keymap layout.

If i found a solution i will post here.

TriForceX, 2019-07-17:
This problem was fixed on the lastest PocketGo CFW v1.1 update.

closed 2019-07-17 by TriForceX