TriForceX/MiyooCFW #48: RTC?

yepyepyep4711, 2019-07-07:
Some games need a Real-Time-Clock. While the bittboy has a clock, unless I'm mistaken, it doesn't survive turning off. Doesn't it survive (keep running) a soft shutdown?

If there is no way to keep the clock running, an option to have the device ask for the time at each start could be an approach.

On a side-note, the current key mapping doesn't work with the clock. you can only increase the value, not decrease it.

TriForceX, 2019-07-08:
The clock you see in menus is fake, theres no internal battery or internal clock in the current hardware to keep the real-time-clock feature working...

closed 2019-07-08 by TriForceX

yepyepyep4711, 2019-07-08:
ouch. This is bad. Can we at least fake it? Say I want to play Pokemon Gold, can I simulate the date and time change by changing the clock before I play. Does this information gets passed to the emulator and the game?

yepyepyep4711, 2019-07-29:
I've found this, which might help people having the same issue http://www.romhacking.net/hacks/3841/