TriForceX/MiyooCFW #49: option to open the selector in the rom folder automatically after boot

yepyepyep4711, 2019-07-09:
see https://github.com/TriForceX/NewBittboyCFW/issues/43#issuecomment-509615452

TriForceX, 2019-07-11:
go main settings -> default path folder

yepyepyep4711, 2019-07-11:
there is only one "settings" dialogue, with language, date and time, etc. It doesn't have a "default path" item

TriForceX, 2019-07-11:
oh, so it doesnt have that added yet (i dont have my console here now). i just checked on LDK/RS-97 because both uses the same system, but the Bittboy/Pocketgo is an older version i think... 

Another alternative is to edit the settings per emulator (press select button on the app), then change the "selector path/folder". tell me if that option is on it.

yepyepyep4711, 2019-07-11:
> oh, so it doesnt have that added yet (i dont have my console here now). i just checked on LDK/RS-97 because both uses the same system, but the Bittboy/Pocketgo is an older version i think...
Any chance we could update to the newer version?

> Another alternative is to edit the settings per emulator (press select button on the app), then change the "selector path/folder". tell me if that option is on it.
Yes, it is there, but it is also already set and works. But what I was asking for was the option to bypass the screen where you have to pick an emulator and go directly to the rom folder after boot. If I understand the way the device works correctly, it would mean opening a "link" like "Gameboy" straight after boot.

It could be called "default emulator", or something

TriForceX, 2019-07-11:
The current system doesnt work like that, but anything can be changed, modified and updated, the source code is available for everyone.

If anyone is interested to make something like that i will left this case opened.

yepyepyep4711, 2019-07-11:
well isn't this a linux system? If so a crontab that opens the desired link should do the trick, but I don't see the relevant folders on boot or on main. A user-friendly solution would obviously need a GUI setting possibility, but to do a proof of concept I'd be happy to try and hack it and see if it works