TriForceX/MiyooCFW #50: repeated crashes with gambatte

yepyepyep4711, 2019-07-10:
hi,

I'm having repeated crashes with gambatte and pokemon yellow. Any way to proceed to try and find out why and how to fix?

TriForceX, 2019-07-11:
which version of pokemon yellow? (EUR, USA, JPN)?
is a random crash? or it happen in a specific case?

yepyepyep4711, 2019-07-11:
the french one. It seems random so far, but the effect has been a white screen. The device was not frozen, I was able to exit the emulator or reset the game. It happened another time with Tetris (W) (V1.1) [!]. That time, the screen didn't turn white, the whole device froze. No reaction from any button. I had to cut the power.

TriForceX, 2019-07-11:
It only happens with that game? have you tried another version of pokemon yellow? also other roms? i had a similar issue with another pokemon game, i was using a bad rom... i asked the Gambatte developer and he doesnt have any idea, maybe theres some incompatibility with something. So i think the rom is the problem.

yepyepyep4711, 2019-07-11:
well these are two different type of crash with two different roms, so no, I can't pin it on the rom.
As I said, any logging we can try and and activate to find out what the problem is?

TriForceX, 2019-07-11:
can you provide the md5 checksum of you conflicted roms?

yepyepyep4711, 2019-07-11:
Sure. 982ED5D2B12A0377EB14BCDC4123744E and 2DF6B439A35E0D511D52FA75C6A7849A

yepyepyep4711, 2019-07-16:
Another crash, Pokemon again, this time the screen froze and the device stopped reacting, like that time with tetris. This would corroborate the theory that this is not rom dependent, as the tetris I'm using is certainly the one most used outside of japan.

yepyepyep4711, 2019-07-17:
I've now had several of these. I also had the graphics turn to garbage will the game would keep playing. Can we *please* debug this? My kid has had to restart from last save three times today. This is making the device look useless. This is *not* a hacked rom we're talking about here. This has to work.

TriForceX, 2019-07-17:
Can you try the lastest Gambatte update? (PocketGo CFW v1.1)

yepyepyep4711, 2019-07-17:
I have a Bittboy, not a PocketGo, so I'll gladly do this when the CFW is next updated

yepyepyep4711, 2019-07-23:
new crash, this time with Donkey Kong (JU) V1.1 [S]. Everything froze, had to cold boot and got the black screen at the first boot as in https://github.com/TriForceX/NewBittboyCFW/issues/52
Maybe it's just the hardware?

TriForceX, 2019-07-23:
Thats weird, no one is experiencing your problems, also the gambatte developer and he doesnt understand, maybe is something outside the emulator, maybe the rom or your firmware. 

Can you try the lastest gambatte update? https://boards.dingoonity.org/gcw-releases/gambatte-13651/

yepyepyep4711, 2019-07-23:
thx for the link. Now I'm trying to figure out how to update :sweat_smile:

yepyepyep4711, 2019-07-23:
ok, I've updated. Time will tell.
"You can read the manual by highlighting the Gambatte icon in Gmenu2x and pressing SELECT. Then choose "Show manual of Gambatte" to read the manual." Is this supposed to work on the Bittboy? I can't seem to find a screen where it does.

yepyepyep4711, 2019-07-23:
#offtopic oh wow, I was about to ask whether correct SGB colors were possible in Gambatte. Turns out, there's a bounty open for that feature https://www.bountysource.com/issues/76534488-incorrect-coloring-on-super-gameboy-enhanced-games-and-super-gameboy-borders

yepyepyep4711, 2019-07-23:
oh, and thanks @TriForceX for pushing me to update Gambatte. I hope this will fix it. It feels snappier in any case and while it has nothing to do with updating it's also how I discovered the default color filter was turned off, and turning it on is so much better for the eyes and the looks :)

TriForceX, 2019-07-23:
Ok, tell us if the update works for you.

PS: Please dont off-topic, more people is following this case (we dont want to spam notifications to everyone), you can add any issue or suggestion in a new topic.

yepyepyep4711, 2019-07-26:
so far only one crash since the update. It is an improvement, at the very least. Let's hope it stays. I'll be gone for a few weeks, so catch y'all after the break :)

TriForceX, 2019-07-26:
Good to hear that, i will close this issue because is not "repeated" anymore for you. If you got a new one please create another issue giving us the console, rom (with md5 checksum) and emulator version.

closed 2019-07-26 by TriForceX

yepyepyep4711, 2019-09-05:
so after a few new crashes I switched machines. No crashes since then. So it seems it was a hardware issue