TriForceX/MiyooCFW #51: is there a chance of switching to retroarch?

yepyepyep4711, 2019-07-11:
it would outsource the keeping up-to-date of the emulators and forward compatibility when the device reaches end of life

TriForceX, 2019-07-11:
Steward already did the tests for PocketGo:
- Retroarch: https://steward-fu.github.io/website/handheld/pocketgo/build_retroarch.htm
- More emulators (PocketGo section): https://steward-fu.github.io/website/handheld.htm

I dont know if anyone already tested the performance on this, but atleast in RS-97 its like 2x~3x slower than normal builds. (remember RS-97 is faster than Bittboy... if runs slowly on RS-97 i cant imagine on Bittboy/PocketGo)

closed 2019-07-11 by TriForceX

yepyepyep4711, 2019-07-11:
ouch. alright, never mind then

gameblabla, 2019-08-29:
Its closed but i just want to reaffirm the speed issue with Retroarch on small handleds.
Direct tests were made and even the likes of GPSP runs way below 60 FPS (something like 30 FPS) on most games.
Everything else is even slower than that so it's really not worth considering and not usable.

yepyepyep4711, 2019-09-05:
thanks for looking into it