TriForceX/MiyooCFW #52: screen stays black at startup, needs cold boot

yepyepyep4711, 2019-07-11:
now and then, when turning the device on, the starting sound will chime, but the screen will stay off. One then needs to turn off and on again. It may be happening if the device went to sleep automatically

yepyepyep4711, 2019-07-16:
It happened again earlier today. I noticed that I could still do things like suspend or turn the screen lighter or darker. I was even able to do a soft shutdown "blind".So it's really some command to switch the display/launcher on that doesn't get issued or crashes randomly, because the device itself is not frozen.

yepyepyep4711, 2019-07-16:
I can reproduce!
If you start the device and hold down all four push buttons, you'll greatly increase your probability of seeing either the screen freeze on NxHope, on some ghost of your last game, or stay black while the leds in the background stay on

TriForceX, 2019-07-17:
This problem happen mostly when the console doesnt shut down properly. In most of cases you need reflash the SD because the partition load gets corrupted.

Also the sleep timeout is not properly implemented on this.

yepyepyep4711, 2019-07-17:
Well as I said I can reproduce this even with a clean shutdown, just by pushing all buttons.

> Also the sleep timeout is not properly implemented on this.
How do you mean that?

TriForceX, 2019-07-17:
I mean the black screen is also caused by the sleep timeout (suspend mode) because is not very well made. Also is caused by a bad console shutdown. And now is caused by the buttons combinations. 

All these things doesn't mean theres a solution, the console itself is not very well made, and there are many ways to "broke it". Some people is trying to fix these issues, but the development is "slow" because there aren't enough people to do it, everyone who's actually working is not affiliated with any company, just normal people. Also we are not getting help from the manufacturers, so all of this is just by trial-and-error.

Thanks for reporting this.

yepyepyep4711, 2019-07-17:
ah, I've always wondered if the people working on this project had anything to do with the bittboy company at all, since there doesn't seem to be an "official" Firmware anywhere.
In any case, many many thanks for all your work. You're making a difference.

gameblabla, 2020-11-20:
Feel free to open it if it's still an issue.

closed 2020-11-20 by gameblabla

yepyepyep4711, 2020-11-20:
well it is, but if it's hardware-based there's not much point in keeping it open, so leave it closed.