TriForceX/MiyooCFW #54: turn led red when charge low

yepyepyep4711, 2019-07-16:
I can only assume what just happened. Mid-game, the device shut down. Since I had been playing for some time, it could be I ran out of juice. After that, it took about five boot cycles to be able to start the machine again (before that, the device froze during boot or displayed endless error messages). Some of the game saves before that (and there had been some, I was playing Wario Land 3 and that game saves regularly) were lost.
Maybe it was another of the crashes I've been having (https://github.com/TriForceX/NewBittboyCFW/issues/50), but it looked different.

In any case, please make use of the fact that we have a power led that can turn red to add the function that it will in fact turn red to signal a low charge to the user. Even better, make it three steps. E.g.: 10% charge, the led turns red, 5%, it flashes red and green, 2%, the device "presses" the R key so that the menu is called up, giving the user a chance to save before the device shuts down (it would also make it impossible to ignore the fact that the device needs to be plugged in)

TriForceX, 2019-07-17:
The battery warning in this console is not properly implemented because the power management circuit on the SoC is bad... Im not sure if is fixable through software.

yepyepyep4711, 2019-07-17:
Sorry, what is the SoC? :sweat_smile:

TriForceX, 2019-07-17:
“SoC” refers to the system-on-chip, an integrated circuit (also known as a "chip")

yepyepyep4711, 2019-07-17:
ah, thx. So what is bad about it (if you don't mind explaining)?

TriForceX, 2019-07-17:
Thats mean you cant expect something to work properly if the manufacturer poorly implemented the battery warning on the system.

If there are any solution on the futher i will post here. Thanks.

akg83, 2023-05-31:
> The battery warning in this console is not properly implemented because the power management circuit on the SoC is bad implemented... Im not sure if is fixable through software.

From what I understood from researching a bit, the battery status is available, only it's awfully not precise so the reading is a "gamble". but could we set the LED to red when the status is like really low? better than nothing.

@TriForceX  or on the flip side, if this can not be implemented at this moment at all, shouldn't this issue be sealed / closed and moved to a discussion thread? so the issue least can be cleaned up and the focus will be given to the open issues that can be worked on, just a suggestion ;-)

Apaczer, 2023-12-25:
IMO duplicate of #106 if it goes for low charge led status. The data corruption shouldn't be an issue now and we added low battery warning a while ago in daemon.

closed 2023-12-25 by Apaczer