TriForceX/MiyooCFW #58: Vertical games in mame get cut off

miwasp, 2019-07-25:
Vertical games like donkey kong or pacman are cut off at the bottom in mame4all. This is even when you select the option "full screen". You can rotate the screen in the mame settings and play it rotated but for some games this is not practical.

TriForceX, 2019-08-04:
You should try **Scale Aspect** or **Scale Aspect Fast** options instead

miwasp, 2019-08-04:
Unless i select one of the rotate options, it doesn‘t matter which option i select, the display is ALWAYS cut off at the bottom.
Try it out:
- Start mame4all
- Select rom dkong.zip (Donkey Kong (US))
- Choose ANY of the non-rotate options

Does it work for you?

miwasp, 2019-08-25:
This is the diff to solve that:

```
miwasp$ diff odx_frontend.cpp odx_frontend.cpp.org 
804c804
<       mame_args[margc]="-bestscale";
---
>       mame_args[margc]="-aspect";
```

Compiled version here: https://www.dropbox.com/sh/er4t0h4t1ax0dvr/AADNUN7NoXjuViYWY7Rvqm1la?dl=0

iamartwo, 2019-08-25:
how can I use your patch @miwasp ?

miwasp, 2019-08-25:
Make a backup of \emus\mame4all\mame4all.dge on the main-partition of your sd-card and replace it with the one you can download via the above link.

closed 2019-09-24 by miwasp

donaggio, 2021-11-29:
Sorry to revive this thread, but the issue still persists with the latest CFW on Powkiddy Q90: I believe that the proposed patch has been already applied (mame4all.dge has the exact same size of the one found in the archive from @miwasp) but I tried it anyway with no luck. Vertical games like 1941, Donkey Kong and many others lack the bottom portion of the screen.