TriForceX/MiyooCFW #59: KeyMapping on Menu and games does not work properly.

VictorCarm, 2019-07-25:
Hello,

I have been installed your CFW on a BittBoy 2.5, and actually I cant navigate into the menu. I tried to do my own keymapping without any knowledge, editing 'input' file into gMenu2x folder, but I only got a correct D-pad Up and Down functionallity. With that solved, I could open a game, but these inputs still working badly into that game (Streets of Rage and Pong, for example).

I reinstalled all multiple times, with multiple versions, including a research for v1 model (in case I was installed a buggy CFW), but I cannot navigate into the menu without shutting down the Bittboy by accident, entering into a config, etc.

Thank you for your time. I can't say I am an expert with these kind of software, so that, all help will be appreciated.

antonio-cpu, 2019-07-31:
The same happens to me. Two days ago I have the console, and I have no way to move through the menus. I have tried different firmware, and I have not succeeded.
Thanks for your help

nikolaosginos, 2019-11-21:
I have the same issue. The buttons don't do what they're supposed to do. I tried v1.3.1. Interestingly, the way the buttons "work" is different between the v2 and v3/v3.5 cfw.

Any idea what might cause this? Or has someone even found a solution meanwhile?

TriForceX, 2019-11-21:
Im checking the problems, because it seems to be present only on some devices... We tried the cfw on all versions before its release, this is a weird case. For now please use the cfw 4.2 (is on the homepage at resources link)

tchaly-bethmaure, 2021-07-14:
Hi @TriForceX,

I have the same issue, what do you mean by cfw 4.2 ? I only see those custom firmwares :
[image]

Regards,
Tchaly

gamer4life1, 2022-03-04:
@tchaly-bethmaure where did you find that?
Thanks

Apaczer, 2023-04-14:
We update miyoo_kbd module as it seems there are hybrid Bittboys with stuff from other consoles.

closed 2023-04-14 by Apaczer