TriForceX/MiyooCFW #60: Neogeo stock at decryption

Levrault, 2019-07-29:
Hi,

I got the following issue on the pocketgo with the firmware 1.1. Whenever I try to play some NeoGeo game, the console tries to decrypting it but get stock a 50% every time. I can't quit this state, I can only power off the console.

Is there something I'm doing wrong? The ROMs were working on my retropie.

jcipen777, 2019-07-31:
I'm stuck too when decrypting metal slug 3,4 and 5, senguko 3.

Levrault, 2019-07-31:
I think I get what the issue is. If the Rom is larger than 40mo, the pocketgo will not decrypt it. I don't know if this is a material issue or not.

TriForceX, 2019-08-04:
Are you using the right roms from MAME 0.128 romset?

Levrault, 2019-08-05:
Hmmm, it's possible that Garou and Metal Slug are all using a more recent MAME. But I thought they just need to be compatible with the good version of FBA?

TriForceX, 2019-08-05:
If you are using GnGeo emulator the roms need to be from MAME 0.128 romset
If you are using FBA-A320 emulator the roms need to be from 0.2.96.86 FBA romset

it is not recommended to use other romset since the emulator probably will not read them properly.

Levrault, 2019-08-05:
Thanks it worked !

closed 2019-08-05 by Levrault