TriForceX/MiyooCFW #62: Considering swap Confirm/Cancel Keys

hu6360567, 2019-08-04:
Hi,
    Thanks for your amazing CFW.
    However, in most asian areas, A is used to confirm and B is cancel. Is there any way to configure them in menus?
    Configurations of some emulators seem to be compiled statically.
    So would you distribute an alternative version with swapped key functions?

TriForceX, 2019-08-04:
Thanks for your message. We are aware of that and yes is possible now due we got an updated source code, but you need to wait first because we need to get this working for all console revisions.

Stay tuned.

gameblabla, 2019-08-12:
I think the best way to fix this would be to reassign the keys kernel side rather than tweaking each emulator and stuff. We shouldn't have to fix up their crap.

doodlewind, 2019-12-11:
I've compiled the kernel, guess that it's something to do with `drivers/input/keyboard/miyoo_kbd.c`. How can we modify the code for swapping?

closed 2021-10-15 by TriForceX