TriForceX/MiyooCFW #63: [GBA] Dragonball Z Gokus Legacy needs savegame type EEPROM

nakotic007, 2019-08-12:
The current gpsp emulator version is not able to change the savegame type. some games as **DBZ legacy of goku** do run on other emulators by changing the savetype to "EEPROM".
Would you guys please add this feature or fix it somehow? 
The error after successfully booting the game says sth like "the hardware you use to run this software is not supported"

example: fix for visual boy advance emu
https://gamefaqs.gamespot.com/gba/519688-dragon-ball-z-the-legacy-of-goku/answers/325658-emulator-problem

gameblabla, 2019-08-12:
I see that the japanese sequel is in the game_config.txt list but not the international version.
Would it be possible to have the MD5 checksum of your rom ?

Also i had updated GPSP but they messed up the CFW, it's just a mess... (though it fixes other games, just not this game yet) :
https://gameblabla.nl/files/ipk/bittboy/gpsp_gameblabla_v3_rumble_mapping.zip

nakotic007, 2019-08-12:
Hi @gameblabla thanks for the response.

These are at least the versions I have tried:

Dragon Ball Z - The Legacy of Goku (E) (M5)
8246b72cb4b6872724836776c123cb55

Dragon Ball Z - The Legacy of Goku II (U)
4e1565aa1f0627c2bf5e51019440cb70

Dragon Ball Z - Buu's Fury (U)
3a74fce97f1ea2b28c2a50ec3df0acee

Best Regards
Nakotic007

gameblabla, 2019-08-12:
Well its strange because i tried Buu's Fury and it works fine here. Are you sure that you have the latest version of GPSP ? Earlier versions had issues with game saves.

nakotic007, 2019-08-13:
Hi @gameblabla 
ok I have tried it again and it seems as if just the first game does not run. 

Dragon Ball Z - The Legacy of Goku (E) (M5)
MD5: 8246b72cb4b6872724836776c123cb55

The other two ran well although they are not mentioned in the game_config.txt.. very strange. How are the mechanics here?

Could you give me game_code and vendor_code of the not running game? I would just add it by myself to the config. How do you extzract these info out of the rom? 

Thank you so much, I really apprechiate your help

gameblabla, 2019-08-13:
I also tried that game on my latest gpsp version and the game works just fine too.
I tried it on my x86 computer though (my bittboy is broke) so no reason to believe it shouldn't work on the bittboy too.

Make sure to update to the latest version i linked above, the CFW does not contain it !
The older versions had issues with some games not booting up.
If you want to force the game to EEPROM saves, here it is :
# Dragon Ball Z - The Legacy of Goku (E)
game_name = DRAGONBALL Z
game_code = ALGP
vender_code = 70
save_type = eeprom

nakotic007, 2019-08-13:
Thanks for your effort, but it still does not run. I have updated the emulator _(also changed section file name "gpsp" to "02-gpsp" as on original CFW)_ and tried out these config entries:

# Dragon Ball Z - The Legacy of Goku (E)
**game_name = DRAGONBALL Z**
game_code = ALGP
vender_code = 70
save_type = eeprom

and 

# Dragon Ball Z - The Legacy of Goku (E)
**game_name = DBZLGCYGOKU2**
game_code = ALGP
vender_code = 70
save_type = eeprom

since the original japanese entry was

# ドラゴンボールZ - THE LEGACY OF GOKU 2 INTERNATIONAL (J)
# Dragon Ball Z - The Legacy of Goku II International (J)
**game_name = DBZLGCYGOKU2**
game_code = ALFJ
vender_code = D9
save_type = eeprom

I adapted the game_name from the japanese version since your proposal was not consistent to the japanese one.

No luck since now :-(

gameblabla, 2019-08-13:
I'll take a look at it once i get my PocketGo but if it still doesn't run despite it, it might be a big in the ARM dynarec. Like Super Monkey Ball, there's not much that could be done. (though i do have a VBA Next port planned next for games like these)

nakotic007, 2019-08-13:
yeah I think this game is also an exception, the compatibility list here says something similar:
https://wiki.maemo.org/GpSP/Game_Compatibility

But nevertheless, much regards for your time. I see forward for the next release, thanks in advance!

nakotic007, 2019-08-27:
Some games make trouble with gpSP .. https://wiki.maemo.org/GpSP/Game_Compatibility

nakotic007, 2019-08-30:
Hey @gameblabla I just tried a few other roms and I just got a version with different MD5 that works with your latest Emulator Version. Thanks! :-)

gameblabla, 2019-08-30:
What's the MD5 of that rom ? Maybe you were trying to play a bad dump.

nakotic007, 2019-11-11:
Works now, I found a Rom that runs well. Sorry for inconvinience

closed 2020-11-16 by gameblabla