TriForceX/MiyooCFW #66: Brightness level issue - CFW 4.1 BittBoy v3

varayoudfederico, 2019-08-14:
I updated my Bittboy v3 to CFW 4.1 and there is an issue with the brightness level of the screen.

**Every time the screen turns off and on again, the brightness level resets to a level that seems to be about 70-80% full brightness.** This happens wheter you restart the device or you suspend it. In previous versions, the brightness level is mantained when you power off the device.

**And also, you have to change the brightness level pressing the buttons once and then release. If you mantain the button pressed, the device crashes.** 

I noticed this because my device has an issue where you can hear a faint high pitch noise every time the screen is not at 100% due to the PWM signal that controls the brightness (by the way, there is a fix for this problem on the official Reddit that would be awesome to implement in future versions of the firmware).

Thanks for you awesome work.

TriForceX, 2019-08-23:
Can you try to update to CFW 4.3 and try again?

TriForceX, 2019-09-03:
This still happens on 4.2

gameblabla, 2020-11-20:
A PR was merged to fix this (hopefully). I'll close this as soon as a new kernel release is made.

Apaczer, 2023-04-14:
Currently the brigthness is controled directly by gmenu2x settings, if you launch app in another way e.g. with ``autoexec.sh`` (or bootlogo) it will use ``daemon`` while reading "/mnt/.backlight.conf" and if no value is provided the default is ``5``

closed 2023-04-14 by Apaczer