TriForceX/MiyooCFW #67: Problem with MAME ROMs

Youpi-38, 2019-08-15:
Hello ;-)
**First of all thank you for the development of these new CFW. (Strongly the following ;-))** 
I have a problem with some MAME ROMs that do not appear in the game list when I open MAMA4ALL (WORLD RALLY for example).  
I downloaded it on several ROM site but nothing helps.  They remain invisible.  
Can you help me ?  
(Sorry for my English, I'm French).

miwasp, 2019-08-15:
mame4all in CFW 1.2 requires the romset version 0.37b5. World Rally is simply not part of that romset.

jcipen777, 2019-08-16:
Play it using fba emulator, i used it

Youpi-38, 2019-08-16:
**@miwasp** 
Hello and thank you for your answer.
OK so there is more than one type of emulator mame4all.
But when you download a ROM, how do you see for what type of emulator it is intended?
Is it possible to modify a ROM for the redefinition "compatible" with this version 0.37b5?
And finally, can you tell me on which site you download your MAME ROMs for this version?

**@jcipen777** 
Hello,
thank you, I just try but the procssus starts and returns directly to the list of games.
I tried four different folders, but unfortunately, same result.
:-(

TriForceX, 2019-10-15:
MAME4ALL uses **Mame 2000** and **Mame 0.37b5** romset.
We cant share roms links, but if you search using these keywords on Google you will get good results.
If your rom is not available in that romset you may use another emulator like FBA-A320

closed 2019-10-18 by TriForceX

Youpi-38, 2019-10-18:
Hello and thank you for your answer.  I will try this.  Thank you ;-).