TriForceX/MiyooCFW #69: PLAYSTATION ROMs

Youpi-38, 2019-08-15:
Hello again, 
I would like to know how to use the PS1 emulator on PocketGo CFW 1.2 once I downloaded the ROM folder in .7z.  
I don’t understand how it works.
Thank-you ;-)

jcipen777, 2019-08-16:
Just extract to your ps1 roms and play it

Youpi-38, 2019-08-16:
Hello and thank you for your answer.
Ok so unzip the file 7z and I get this:

**- (PSX NTSC) Bomberman World.ccd**
**- (PSX NTSC) Bomberman World.cue**
**- (PSX NTSC) Bomberman World.img.ecm**
**- (PSX NTSC) Bomberman World.sub**

I put the four files in the ROM -> PS1 folder?

;-)

jcipen777, 2019-08-16:
Just download bin and cue file, put it on ps 1 file.

Youpi-38, 2019-08-16:
Nothing works ..... 
In **Bomberman** I only have the **.cue** file (in addition to the other three files), which is invisible if I put it alone as I do not have a **.bin** file.  
If I put the fourth files it does not work either.  
I tried with **DRIVER**, and in this folder I only have the **.bin.ecm** file that does not work either.

gameblabla, 2019-08-16:
ECM is not supported, please decompress it with the unecm tool before attempting to play it again.
Also check the cue file and see if it says (PSX NTSC) Bomberman World.img.

That's why i tell people to rip their own games...

nakotic007, 2019-08-28:
In case you can read german: http://powerpi.de/forum/viewtopic.php?t=685
this tutorial helped me a lot when I set up my raspberry back then

gameblabla, 2019-08-29:
Well i'm considering this done.
The ECM format will never be supported and shouldn't be preferred due to memory constraints. Decompress them with a tool like unecm and you should be good to go.
If you have another issue with the PSX emulator, then just open another issue. Thanks

closed 2019-08-29 by gameblabla

Youpi-38, 2019-08-29:
Hello,
 sorry I did not answer because in the meantime I went on vacation and I do not have a PC.  
I will try to see that Sunday or Monday and you will hold to the capturant. 
Thanks for your help.