TriForceX/MiyooCFW #70: Problem(?) with 6 button gamepad config in Picodrive

kuzmapunk, 2019-08-22:
Hello! :)
Every time when I open rom in picodrive I must save global or local configuration to use 6 button configuration. How can I fix this problem? Or maybe it’s not a problem and I do something wrong?
Thanx for help in advance ;)

gameblabla, 2019-08-23:
It is working the way it is intended. Some games have issues with the 3 buttons mode which is why it's not set as such by default. If you want to use the 6 buttons mode by default, you can simply just save globally your config. Or use it per game with the local configuration for each game.

kuzmapunk, 2019-08-23:
Thank you for reply. I know it. But In picodrive it’s look like game on start load 3 button config by default in any case. And if I go to gamepad settings and select “save config” - all 6 buttons work fine :) I’ve try it on pocketgo cfw v1.2, games: Beyond oasis, Mortal Kombat 3 Ultimate and Comics Zone. I think it’s work similar for all games.

gameblabla, 2019-08-23:
I'll look into porting Picodrive again to the buttboy after this PR is merged :
https://github.com/notaz/picodrive/pull/100

It doesn't happen to me strangely enough, looks like the default config doesn't work properly.

closed 2021-10-15 by TriForceX