TriForceX/MiyooCFW #72: R and L buttons backwards in File select

berserkfreak, 2019-08-24:
Since update 1.2 R and L are backwards in File select R scrolling down L Scrolling up.

berserkfreak, 2019-08-25:
Sorry I mean the Reverse L scrolls down R scrolls up

nakotic007, 2019-08-27:
yep, very annoying since it was solved with previous CFW

nicosensei, 2019-11-13:
There is an easy fix: you can edit _/mnt/gmenu2x/input.conf_, you can do it directly on the device using WizWrite, or on a PC:

Be sure to have the 2 following lines as below (all other lines remaining unchanged): 

```
section_prev=keyboard,9
section_next=keyboard,8
```

Here's the full file:

```
up=keyboard,273
down=keyboard,274
left=keyboard,276
right=keyboard,275
modifier=keyboard,304
confirm=keyboard,308
cancel=keyboard,306
manual=keyboard,32
section_prev=keyboard,9
section_next=keyboard,8
pagedown=keyboard,8
pageup=keyboard,9
settings=keyboard,13
menu=keyboard,27
power=keyboard,305
dec=keyboard,9
inc=keyboard,8
```

eknives, 2020-12-02:
I modified input.conf like this and I have lost the Reset function, It is already lost, I tried another SD, I tried CFW, OFW, and Reset function is lost in my Powkiddy Q90.

How can I fix it?

Apaczer, 2023-03-11:
update config on all builds: 
R1 scrolling DOWN 
L1 Scrolling UP.

closed 2023-03-11 by Apaczer