TriForceX/MiyooCFW #75: trading pokemons

yepyepyep4711, 2019-09-28:
I know it's not strictly a Bittboy question, but I thought I'd get good pointers here, as I can't find the answer anywhere.
I'm using Gambatte on the Bittboy to play Pokemon Yellow and would like to trade  with another player. Can this be done? Since it can't be done with the Bittboy itself (that I know of), can Gambatte saved states or game saves be imported in another emulator (BGB on pc, or anything running within Retroarch) to then link and trade? Obviously, this would need to work on the way back too, so that the traded Pokemons find their way back to the Bittboy.

I'm thankful for any advice

gameblabla, 2019-09-28:
You could use ingame saves to do that i guess. The game saves (not save states) are stored in their own .gambatte directory if i am not mistaken...
If you are asking for two player support between two bittboys... Sorry but without wi-fi or something like it on the Bittboy available, this is simply not possible.

yepyepyep4711, 2019-09-28:
> You could use ingame saves to do that i guess. The game saves (not save states) are stored in their own .gambatte directory if i am not mistaken...
That is correct. But how to do it? Are .sav compatible through all emulators? So for example two instances of BGB on PCs, each running roms and .sav from the bittboy in different folders then linking over IP?

gameblabla, 2019-09-28:
Yes, the .sav is simply the raw dump of the SRAM, it's compatible with all emulators and even the real hardware. The only thing not compatible are save states.

Cissoubaka, 2019-10-13:
There is GB emulator on PC (tgb dual and sameboy) that can load 2 differents games and exchange pokemon for example

Valnjes, 2019-10-30:
But You dont need that - you can just use GameShark to enable "catching pokemon from other trainers", so you can catch all pokemon you need :) 
(i know, im a bad person for that, but hey, sometimes i want this Gengar without trading)

gameblabla, 2020-11-20:
Someone would need to port nTrade Advanced to SDL 1.2 :
https://www.omnimaga.org/ti-nspire-projects/ntrade-an-on-calc-trading-program-for-3-gen-pokemon-games/

Probably by making an SDL 1.2 for it.

closed 2021-10-15 by TriForceX