TriForceX/MiyooCFW #77: magnavox odyssey 2 emulator

Cibomatto2002, 2019-10-09:
In the future could we see a Magnavox Odyssey 2 emulator on the Pocket go?
 This emulator is old but better than nothing. 

http://o2em.sourceforge.net/

reset-selectgame-1-9, 2020-10-02:
There was one for the Dingoo A320, don't know if it's compatible:
https://dl.openhandhelds.org/cgi-bin/dingoo.cgi?0%2C0%2C0%2C0%2C27%2C680%2C0%2C0%2C0%2C42%2C721%2C0%2C0%2C0%2C46%2C695%2C0%2C0%2C0%2C71%2C281%2C0%2C0%2C0%2C30%2C174%2C0%2C0%2C0%2C71%2C68,0,0,0,71,106

G7200, 2020-10-17:
Here's another Odyssey2/Videopac emu i've found. But this one comes in a opk:
https://boards.dingoonity.org/gcw-zero-emulation/port-requestpossibility-o2em/

salvacam, 2021-03-06:
I have tried to port it but it uses a library, Allegro, which I cannot use with the toolchain. I don't know how to compile with that library

Cibomatto2002, 2021-03-07:
Ok thank you

salvacam, 2021-03-20:
I have compiled the code inside this opk
https://drive.google.com/file/d/0BwYi4RGX-HSaYjlRMkh3UU5HcFU/view
But I do not pass a screen that says "select game"
I got it from this thread
https://boards.dingoonity.org/gcw-zero-emulation/port-requestpossibility-o2em/
Hopefully someone with more knowledge in C + can try to get it to work

G7200, 2021-03-20:
Wow, that's a good sign.
SELECT GAME is the boot screen, you need to insert a game and mostly start it by pressing a number.
You can check all the game manuals here:
https://videopac.ch/

closed 2021-10-15 by TriForceX