TriForceX/MiyooCFW #80: Brick the firmware

fcastellanosKreios, 2019-10-14:
Hi, the cfw works great, but I have notice that when I enter: select + TA at the main menu, the screen goes black, after that the cfw doesn't respond. The worst is that after I switch off and on, the "unclear shutdown" message appears, the bittboy splash screen and then noting, a black screen and the cfw just don’t boot. 
I am using the bittboy V3.5 and the cfw V1.3.1
Thanks!!

Valnjes, 2019-10-27:
I have the same issue. Plus the PocketGO vibrates every few minutes.
Im flashing the 1.32 and hope it works again.

Valnjes, 2019-10-28:
1.32 Fixed the issue on my PocketGO.

TriForceX, 2019-10-28:
The firmware doesnt get bricked... is just the brightness that is goes set to 0, the volume & brightness key combos changed on the lastest firmware, so is recommended to use the R + DPAD (see wiki [here](https://github.com/TriForceX/MiyooCFW/wiki#console-hotkeys))

Of course you get an "unclean shutdown" because you turned off the console from power switch... if goes black just use again SELECT + TB or R+RIGHT

closed 2019-10-30 by TriForceX

homeswitcher, 2022-03-19:
OMG THANKS SO MUCH MY BRIGHTNESS WAS 0 I WAS ABOUT TO HAVE A MENTAL BREAKDOWN