TriForceX/MiyooCFW #82: GBA problem

RetroLover2030, 2019-11-09:
Some GBA games doesn't work properly. I think it's not the roms but emulator. It's like a blurry gameplay or something. Can you fix this? Thanks.

gameblabla, 2019-11-09:
This comment doesn't make much sense. On the other hand, you complain that some games don't work properly. (which is true. for example, Super Monkey Ball & Steel Empire don't work)

Then you complain about the blurriness of the gameplay ?
If your issue is the upscaler, then you can change it in the graphical options in the menu and change the upscaler to either unscaled (240x160 in a 320x240 window, native resolution) or choose to keep the aspect ratio.

RetroLover2030, 2019-11-09:
> This comment doesn't make much sense. On the other hand, you complain that some games don't work properly. (which is true. for example, Super Monkey Ball & Steel Empire don't work)
>
> Then you [...]

closed 2019-11-09 by RetroLover2030