TriForceX/MiyooCFW #84: Best practice for debugging

doodlewind, 2019-11-30:
Hi, I'm a developer interested in writing apps on Miyoo (PocketGo), and I've created a Docker template for much simpler toolchain setup:

https://github.com/doodlewind/MiyooSDK

For now I've got SDL Hello World running properly, but I'm wondering what's the best practice debugging such a SDL app on Miyoo? I have no idea how to setup a SSH connection or using GDB on the device. It would be very frustrating if I have to write debug log to SD card and manually take it out for inspecting. What't your recommendation? Thanks.

doodlewind, 2019-12-09:
We can compile an ARM GDB on Miyoo and perform remote debug through UART serial console with minicom. The UART port should be soldered according to [Steward's Guide](https://steward-fu.github.io/website/handheld/miyoo/uart.htm). However this only allows one-way logging without input support on PC side, since the RX port on Miyoo is occupied by the audio jack. To solve this, a modified kernel build is required.

However, we can make a simple workaround if we only need to see the `printf` app logs. Just modify `/etc/inittab` in rootfs:

```
# tty1::respawn:/etc/main
console::respawn:/etc/main
```

This will redirect the app debug log to UART port.

If you keep this issue open, I may post other discovery here :)

doodlewind, 2019-12-09:
I've compiled the kernel. Just unzip and replace this file in `boot` partition, which disables audio support, making UART RX usable:

[suniv-f1c500s-miyoo.dtb.zip](https://github.com/TriForceX/MiyooCFW/files/3940008/suniv-f1c500s-miyoo.dtb.zip)

Then we can login the serial console with username "root" and no password. Using GDB and other debug utils on the device should be straightforward now.

doodlewind, 2019-12-13:
The kernel build is also ready, just replace the `zImage` to enable UART debug:

[zImage-debug-ab-revert.zip](https://github.com/TriForceX/MiyooCFW/files/3958773/zImage-debug-ab-revert.zip)

TriForceX, 2020-10-09:
Sorry for the late reply, looks very good, we also are looking for better ways to automatize everything. If you can join our discord to have a talk it should be nice! https://discord.gznetwork.com

closed 2021-10-15 by TriForceX