TriForceX/MiyooCFW #89: Build the kernel for Bittboy v3.5 support

asypost, 2019-12-19:
I build the kernel use the **buildroot** inside this project and found that the kernel can't bring up my Bittboy v3.5.The screen stuck at **NX Hope** can't go any further.So I want to know is this project support Bittyboy v3.5 yet? And if it support it how the process could be to build from scratch?
Thanks in advance.

TriForceX, 2020-10-09:
Are you sure your console is the bb3.5 version? have you tried the latest cfw for bb3.5 before compile it for bittboy?

asypost, 2020-10-10:
@TriForceX Yeah.The official img file works fine on it.

TriForceX, 2020-10-16:
are you setting it MIYOO_VER for proper version?
2 = bittboy v2 and bittboy v2.5
3 = bittboy v3 and bittboy v3.5
4 = pocketgo, powkiddy q90 and v90
atleast in 
**drivers/input/keyboard/miyoo_kbd.c**
and 
**drivers/video/fbdev/r61520fb.c**
maybe the problem youre having is in the screendriver
miyoo_kbd reads it from the defconfig now, but we may add that for the screendriver too... or you can wait for some updates we are working on, because some code is a littlebit messy

gameblabla, 2020-11-20:
Feel free to open it again if you run into some issues as it also works fine for me.

closed 2020-11-20 by gameblabla

hex55, 2021-01-04:
I have the same question, and I modify the MIYOO_VER, the screen stuck at NX Hope can't go any further remain. So can anyone help me?

Ivening, 2021-11-16:
I have the same issue during boot - do not know what am I doing wrong...
I try to compile all of this for LicheePi Nano
I follow the instruction https://github.com/TriForceX/MiyooCFW/wiki/Build-Source
U-boot starts, dtb file loads, but Kernel does not start.
Any other kernel compiled for LicheePi Nano boots without problems...
Tried to change MIYOO_VER=3 but, unfortunately, no use...
By the way - what does "0" mean for this parameter in drivers/video/fbdev/r61520fb.c and "1" for drivers/input/keyboard/miyoo_kbd.c ?
In description above there are no 0 or 1, but only 2, 3, 4
Any help would be very very very appreciated!!!

TriForceX, 2021-11-16:
@Ivening i recommend you to use the upcoming cfw structure described here https://github.com/MiyooCFW/mainboot

that repo basically is how the next cfw will work (an universal image) so the console will be defined through console.cfg

so you should generate your own kernel/drivers and place in the right folder, make your image and test. see how the other devices are made, usually the instructions to manage the screen/keyboard version are under the mainboot scripts, tho do that the device uses the miyooctl app to configure each version, so is not needed to configure through the compiled kernel.

Ivening, 2021-11-17:
Dear @TriForceX, thank you very much for the answer!
I will make a try with a https://github.com/MiyooCFW/mainboot
Is there any communication way with you except issues on GitHub? (mail, chat or something else).

TriForceX, 2021-11-17:
We use Discord chat https://discord.gznetwork.com