TriForceX/MiyooCFW #91: NeoGeo cant run a game

TryEka, 2020-02-10:
I update cfw for PocketGo

And it won't run NeoGeo games

Thanks

TriForceX, 2020-03-12:
You need to place BIOS file neogeo.zip (around 820kb) in '/roms/neogeo' along with games.

Roms set/games need to be from the MAME 0.128 romset.

xXS4ALLx, 2020-05-01:
For those that care or just curious, below is what I posted on Discord a while back before PocketGo was released, so the below is what I posted but updated to include some info for PocketGo.

I managed to figure out the what files should be in the neogeo.zip file that GnGeo needs, keeping the zip file organised:

000-lo.lo
sfix.sfx
usa_2slt.bin <= USA Bios
vs-bios.rom <= Japan Bios
sp-s2.sp1 <= Europe Bios

By default GnGeo looks for the Europe Bios, if you want to change the Bios region in GnGeo, do the following:
- Text edit the following file: SD CARD\.gngeo\gngeorc
- Depending what country you want to use, add one of the following line and save:
country usa
country japan

No need for Europe as GnGeo uses Europe as default ,for those who want to play NeoGeo games in Japanese, this is very handy.

Unibios:
If you want to use the Unibios, simply download the latest version of Unibios, current is v4 and place the "uni-bios.rom" file in the same directory as your NeoGeo roms.
- Text edit the following file: SD CARD\.gngeo\gngeorc
- Add the following line and save:
system unibios
- To edit the Region and Mode for Unibios, press the following buttons "B + A + TA" at the same time when you see the Unibios boot screen right before the game boots.
- Here are the button layout within the Unibios settings: PocketGo buttons in brackets ()
A = A (B)
B = TA (A)
C = B (Y)
D = TB (X)

Unibios Issues:
Current issue is that when using the Unibios, the bios DOES NOT recognise the BittBoy/PocketGo's SELECT button as the COIN button, therefore you will NOT be able to insert coins if you set the Unibois to ARACDE MODE, however in CONSOLE MODE, most games will not require to insert coins as it will have pre set credits or free play, issue might be that the Unibios is using a different KEY mapping for the COIN button.
Only real benefit of the Unibios at the moment is Fatal Fury 3 does not boot with USA/JAP/EUR bios, only boots in the Unibios.

General Notes:
Yes it's time consuming to edit the "gngeorc" file on your PC ever time you want to change something, but what you can do to save this step is to:
- Create different "gngeorc" files on your PC.
- Save them on your SD CARD somewhere, create folders and name then accordingly to the what ever the "gngeorc"  file options used.
- Then use the "commander" app to copy the "gngeorc" file to ".gngeo" folder, overwriting the old file whenever you want to change something.

Alernative method to booting into Unibios:
- On your PC, navigate to the following folder: SD CARD\gmenu2x\sections\emulators
- Find your current GnGeo Emulator settings file, should be something like: xx-gngeo
- Create a copy of this file and rename it: Important notes, the way the emulators are organised on your BittBoy is that it uses a numbering system starting with 01-xx.
Example:
If your current GnGeo emualtor is 11-gngeo, rename the new copy to 12-gngeo, in doing so it will keep the GnGeo together in the emulator list.
Now whatever emulator you have that was 12-xx, you will need to rename this to 13-xx, this needs to be done to all the rest of the files as well.
- Text edit the xx-gngeo file that you have just duplicated:
Rename the title to something you like, below is how I set mine:
title=NEOGEO (Unibois)

Add the following to LINE 4:
params=--system=unibios

So it should look something like this:
title=NEOGEO (Unibois)
description=GnGeo Emulator
exec=/mnt/emus/gngeo/gngeo
params=--system=unibios
clock=702 or 732 or 798 <= See notes below

Notes:
If you have BittBoy, you can change the clock to 798 to help improve the overall performance.
If you have PocketGo, you can try 798, if it's unstable (system crash) change to 732 instead.
Noticed on some PocketGo units, using 798 is fine while others is not stable.

- This will boot GnGeo emulator using the Unibios, this method does not require you to have the "system unibios" entry in the "gngeorc" file.
- You can skip this step and do this all on the BittBoy by manually adding emulator and settings, your new emulator entry will not be sorted and will be at the bottom of the list.

closed 2021-10-15 by TriForceX