TriForceX/MiyooCFW #93: Next Release To-Do Stuff

TriForceX, 2020-03-06:
This is the to-do list for the next release (v1.3.3)
Feel free to comment what things we should update to consider it in a next release.

[x] Update boot animation for Powkiddy Q90 devices
[x] Update boot logo (NXHope) to MiyooCFW logo
[] Update battery driver to allow more parameters to be used (like the charger status)
[] Fix text overflow on file explorer when screenshot preview is enabled
[] Add OPK support

jaimebueno, 2020-04-19:
Add amstrad emulator please 🙏🙏

xXS4ALLx, 2020-05-01:
Add C64 emulator, why has no one ported this over yet?

m45t3r, 2020-05-04:
I did another port of PocketSNES: https://github.com/m45t3r/pocketsnes/releases/tag/v2020.05.03-pgo. I fixed the incorrect mapping of the functions, removed some unused features and did some optimizations (PGO mainly).

This may be interesting to be included in the next version of CFW instead of the current version that we have (that has incorrect mapping of the buttons). There is also the [source code available](https://github.com/m45t3r/pocketsnes) in this one, so if you want to build it from source you're welcome.

m45t3r, 2020-05-09:
So I also decided to (re)port `snes9x4d`: https://github.com/m45t3r/snes9x4d-miyoo

There is tons of optimizations in this port, bought from other Snes9x forks like https://github.com/drowsnug95/snes9x4d-rs90 and https://github.com/RetroPie/ARMSNES-libretro, support to billinear filtering and mono sound (this last one is somewhat broken, but it works well enough and makes things faster). So it may be interesting to also include this version instead of the current one in CFW.

TriForceX, 2020-05-09:
@m45t3r great! ive updated the main list and the to-do list to be added in a futher cfw release. thanks!

m45t3r, 2020-05-10:
Some people asked in `/r/bittboy` for a NeoGeo Pocket emulator. I found one made by @gameblabla, fixed inputs, changed some compile flags (PocketGo needs frameskip or it will run too slow) and compiled it with PGO. Here we go: https://github.com/m45t3r/NGPCemu-miyoo/releases/tag/v2020.05.10

Maybe it would be better to ask @gameblabla if he prefers to his version instead though, [since the repository](https://github.com/gameblabla/NGPCemu) had almost everything needed to do a compilation. But I didn't change the credits or anything, all kudos to him.

willtsan, 2020-05-28:
@m45t3r I checked on yout NGP emu and it runs very slow right now, maybe you can include frameskip in options?

m45t3r, 2020-05-28:
@willtsan It is already compiled with frameskip, and it is basically always on. The core is not really optimized so yeah, this is as fast as we can get.

@gameblabla maybe have more ideas on how to optimize it, I basically took his code and compiled anyway.

bardeci, 2020-06-01:
My fork of Gambatte is now called "Gambatte-DMS". I released a new build today: https://github.com/bardeci/dot-matrix-simulator/releases/latest

nl255, 2020-06-16:
The version of SMS Plus GX in the current firmware has broken save state support and doesn't save the button remapping, is this something that could be fixed by updating the emulator?

gameblabla, 2020-06-17:
My last build of SMS Plus GX didn't have any of these issues as far as i recall, but i could make a newer build if needed.

nl255, 2020-06-17:
The one included with the 1.3.2 CFW for the pocketgo does, at least with Ghostbusters.  To duplicate just save on the title screen then go into the name entry screen and try to load the state and it will crash the emulator. Or save on the city map screen and then try to load the state while you are driving.

Your older build available here, https://gameblabla.nl/files/ipk/bittboy/ , works better though loading states (especially one made on the city map screen that is loaded while you are on the driving part) will randomly crash the emulator.

gameblabla, 2020-06-17:
Yeah there's that bug about save state that was reported to me by the libretro folk. I had released another build on Discord but i don't think they used that either. Your build doesn't have profiling so it will run slightly slower. I'll take the time to update some of my older releases

nl255, 2020-06-18:
That would be great.  As for using outdated emulators they are also using an older gpsp that doesn't even have button mapping support (there are menu options for button mapping and analog stick mapping but they just go to an empty screen with back as the only option).  I replaced it with your gpsp_gameblabla_v3_rumble_mapping.zip and it seems to work fine though I only did a little testing with Mario World.

gameblabla, 2020-06-20:
@nl255 https://github.com/bittboy/gpsp For a while, the CFW was provided only with an older and broken version.

nl255, 2020-06-21:
@gameblabla For the 1.3.2 cfw for the pocketgo v1 that is still the case.

Kayoo80, 2020-07-06:
I'm absolutly not sure about the Game&Watch emulator by Steward. He released it for RG350 but certainly not for Bittboy / Pocket GO.

TriForceX, 2020-07-06:
> I'm absolutly not sure about the Game&Watch emulator by Steward. He released it for RG350 but certainly not for Bittboy / Pocket GO.

https://youtube.com/watch?v=5Vj_iqSZFKk

Kayoo80, 2020-07-07:
Does domaine know where to download it?

https://youtube.com/watch?v=5Vj_iqSZFKk

actraiser30, 2020-07-11:
Can you add the TV-Out back to Bitboy 3.5 ??

And a ZX Spectrum, Amstrad CPC, C64 and VirtualBoy emulator, also openlara (tombraider)  eduke32 (dukenuken3D) and Descent to ports ??

Thanks

reset-selectgame-1-9, 2020-10-13:
Please add a Videopac/Odyssey2 & Intellivision emulator!

sertiana, 2020-10-26:
I agree with the OPK support! 🤗

lucianogardim, 2020-11-22:
Why the status battery not exists?

asypost, 2020-11-26:
ADD SDL2 Support please, lots of indie games use SDL2.

gameblabla, 2020-11-26:
Forget about SDL2. SDL 1.2 is slow as it is on this platform, and besides SDL2 does not currently support fbdev nor does the existing patch for it support framebuffer support in software in SDL2.
Also most of these "indie" games that use SDL2 are also using OpenGL, and this thing has no gpu inside anyway.
(assuming of course, that they are even open source)

freq-mod, 2020-12-25:
Hello. There exists a [greatly updated fork of PicoDrive](https://github.com/irixxxx/picodrive) with massive improvements in 32X department and emulation accuracy. There are .opk packages of this fork for OpenDingux MIPS devices, but would it be possible to support it here as well?

gameblabla, 2020-12-26:
i'm aware of this fork and there's no reason it can't work on the Bittboy yes. A straight compile should be sufficient.

gameblabla, 2020-12-26:
[PicoDrive_bittboy_JIT_32X.elf.zip](https://github.com/TriForceX/MiyooCFW/files/5742904/PicoDrive_bittboy_JIT_32X.elf.zip)
Here's a build for the Bittboy of irixxx's picodrive. This was untested though but aside from Cyclone not liking the console, this should work fine.

freq-mod, 2020-12-26:
@gameblabla thanks, but unfortunately it doesn't work. Upon starting, it crashes to main menu. Tried both via explorer application and ROM load, no results.

gameblabla, 2020-12-26:
What about running the executable itself without trying to load a game, does it still boot at least ? (there's a menu in picodrive)

freq-mod, 2020-12-26:
>What about running the executable itself without trying to load a game, does it still boot at least ? (there's a menu in picodrive)

No, it doesn't. It crashes instantly, I don't get to seeing picodrive's menu.

freq-mod, 2020-12-26:
Tried also using built-in st terminal to run executable. It doesn't work, displaying only the following message:
`plat_sdl: using 320x240 as fullscreen resolution`

Then it locks.

gameblabla, 2020-12-28:
Well i suppose it would need to be properly debugged. Too bad that doing so requires the audio to be disable temporarily : (

nl255, 2020-12-28:
I would like to recommend that the CPU be underclocked as low as practical when the system is put in suspend mode if that isn't something that is already done.

freq-mod, 2020-12-28:
>Well i suppose it would need to be properly debugged. Too bad that doing so requires the audio to be disable temporarily : (

Please do so, PicoDrive really needs to be upgraded with proper 32X support.

freq-mod, 2021-01-03:
Hello. I have noticed @irixxxx's Picodrive was updated with fixes for bittboy related systems. While I tried to build it myself, I couldn't figure out everything, would it be possible to post .elf build of Picodrive?

gameblabla, 2021-01-06:
Does this work better for you ?
I'm surprised you would even get those kind of crashes. (But better than just crashing before it even gets to the menu, i suppose)
[PicoDrive_bittboy.zip](https://github.com/TriForceX/MiyooCFW/files/5778558/PicoDrive_bittboy.zip)

freq-mod, 2021-01-07:
>Does this work better for you ?

It doesn't, it's back to crashing at startup.

irixxxx, 2021-01-08:
Unfortunately I can't help you much with this. I just don't have the hardware. I did some work to fix errors and warnings in the build process, but only to the point of cleanly compiling. Without hardware, or at least someone having the necessary knowledge to compile the source and debug on the device (and willing to spend some time collaborating with me on this) I really can't do much.

gameblabla, 2021-01-11:
I recall Steward-fu having issues with making Cyclone 68k work on the bittboy but i'm not sure how he managed to fix it or if he just ended up using the C interpreter...
https://github.com/steward-fu/miyoo/releases/download/v1.2/picodrive.7z

It seems that he did end up not using cyclone it seems. (he added a miyoo target and in there, optimize_arm920 is not even turned on)

gameblabla, 2021-01-11:
[PicoDrive_bittboy_fame.zip](https://github.com/TriForceX/MiyooCFW/files/5795497/PicoDrive_bittboy_fame.zip)
Here's a build that doesn't use Cyclone but Fame instead. This should run ok (but a bit slower than Cyclone. To be honest though, the original version was also using fame)

freq-mod, 2021-01-11:
I really don't know how, but this thing **also refuses to start**, crashing to main menu just after executing, no matter if I run it from main menu or file explorer
I also tried deleting config file from .picodrive, didn't help...

gameblabla, 2021-01-11:
Well it does sound like it's not an issue with Cyclone or whatever so as he said, i will need to run this under valgrind or so.

joejoejoejoejoejoejoejoejoejoejoejoe, 2021-01-22:
Yeah, I like the .opk idea, maybe another GBA emulator as a option, like reGBA

gameblabla, 2021-01-22:
> Yeah, I like the .opk idea, maybe another GBA emulator as a option, like reGBA

For the 3rd time again ReGBA cannot be ported to any ARM or x86 devices, reGBA is strickly MIPS specific due to the recompiler only supporting MIPS32. Besides, it still wouldn't fix most of the game breaking issues aside from Golden Sun and a few others. Games like Kingdom Hearts, Steel empire etc... still won't work on ReGBA.

But yes as for OPK, i paused the work on my CFW myself but it's not hard to get working. In fact, it could even be added back to the firmware. It's just that we will need to go back to Gmenu2x and repackage the apps and even fix some of those, which is why we still haven't done it. (and probably never will)

dosuserx, 2021-05-04:
powkiddy v90 request: key bindings for the menus of fceux, PocketSNES, gbsp, ohboy and possibly others the action and cancel butons are reverse. if they could be unified it would make the cfw really feel polished. or maybe the user could set the emulator control buttons to fix themselves.

Vantoco, 2021-05-06:
Get improved mame working please !!! :D

tylibra2, 2021-08-15:
It could support connecting Pocketgo to PC with USB cable, without needing an SD card reader. Being able to add .opk files would be a good thing if you can add other emulators or ports if you don't have for Pocketgo. And av out would also be cool for some people like me. I would love to one day test the av out and play on TV!