TriForceX/MiyooCFW #94: Mplayer can only play a file once

horomancer, 2020-03-12:
I added some mp3 files to test the media player. When I access the files through Mplayer, I get a blank screen with the file playing, the only controls I have are Y to pause/start the file and START to exit Mplayer. If I try to start the file again though Mplayer, I see the loading screen, then I am kicked back out to the Apps menu.I can load another file without issue, but only once and then this happens again.
I am using CWF 1.3.2, I remember trying the Mplayer on stock firmware and it having many menus and features which I do no have now.

TriForceX, 2020-03-12:
Which console are you using? (BittBoy/PocketGo/PowkiddyQ90)
Can you share the mp3 file with the issue? (Here on our Discord chat)

horomancer, 2020-03-12:
I am using a PocketGO with CWF 1.3.2
I do no not have access to the mp3 currently, but can confirm they have been used for years on my home laptop without issue. There was no problem or distortion when playing the files for the first time, only when I exited and tried to play the files again.
This issue did no change after rebooting the system.

TriForceX, 2020-03-12:
Some MP3 files has different bitrates and codecs, so maybe Mplayer doesnt recognize properly, i hope you can get the MP3 to do a proper test in the console.

For now we will try the current Mplayer (and the older version) with some MP3 files we have to test.

horomancer, 2020-03-12:
I will have access to those exact MP3s as well as other formats (mostly .ogg) later in the day. I'll test them and post the exact files here. Would there be any other info helpful in trouble shooting this? Maybe an md5sum of the Mplayer file itself?

horomancer, 2020-03-13:
Here are a few of the files that I have attempted-

[testfiles.zip](https://github.com/TriForceX/MiyooCFW/files/4327958/testfiles.zip)

I also noticed that for these files there where additional hidden entries in their respective media folders-

[hiddenfiles.zip](https://github.com/TriForceX/MiyooCFW/files/4327962/hiddenfiles.zip)

I assume this as something to do with the issue
I played the files in testfiles.zip on my laptop with no issue.

mirkokg, 2020-07-20:
If you open file manager and go to the folder where your media files are located, you will find files with same name as your media files prefixed with dot character, delete them and play your files again. mplayer is keeping current playing position in those files, but when you play 'till the end it won't play because player is closing at the end of the file.

gameblabla, 2020-07-20:
That's because its a video player, not an MP3/WAV/M4A player or whatever. Something like GMU would be more suitable for that purpose. Unless of course you were thinking of binge watching multiple videos at a time, which is seriously going to be troublesome for the bittboy.

mirkokg, 2020-07-24:
@gameblabla Depends which codec is used :P It can playback MPEG-2 video without trouble, my little one loves watching cartoons on it :)

symbios24, 2020-07-31:
What about replace the useless Mplayer with VLC arm version than plays everything

bnickodemus, 2020-09-19:
What is problem?

If you quit mplayer then it saves a .resume file
Then when you try to load a file (mp3 or mpeg-2) and you get error line 16: syntax error: unexpected "elif"
This is a problem with bash script, elif is not bash command. bash uses if, then, else, fi structure

How to fix?
modify /mnt/apps/mplayer/run.sh to:

if [ "$EXIT" == "Exiting... (Quit)" ]; then
	#echo "Quit"
	rm "$RESUME"
fi
if [ "$EXIT" == "Exiting... (End of file)" ]; then
	#echo "End"
	rm "$RESUME"
fi

now resume file will always be removed.
see changelog "MPLAYER" changes

bnickodemus, 2020-09-19:
I had the same problem @horomancer 
see my solution above.