TriForceX/MiyooCFW #98: ScummVM anyone?

BryMD, 2020-05-21:
Hi guys! Who do you think I have to bother in order to get ScummVM ported to the original PocketGo? Would love to do it myself, but have literarily zero knowledge of linux programming. But if you ever need help with Mac OS (Classic and X) scripting/terminal/GUI, I'm your guy ;)

BryMD, 2020-05-21:
PS: A big shout-out to everyone involved with the CFW! I know its oftentimes an ungrateful endeavour with lots of unmentioned and/or misplaced gratitude. So hats off to you!

BryMD, 2021-02-14:
Freakin' AWESOME job on the port of ScummVM, salvacam! Showing my appreciation here as I can't find a way to comment on the fork itself.

Have designed an alternative icon for the FontiGrid theme, as the default one broke with the design convention. The pixel art is inspired by the 1986 Apple Mouse, and has been made on Mac OS 7.5.5 for good measure :) Feel free to use it!

[image]

RichardNobel, 2021-02-15:
Sounds good @BryMD ! Would that ScummVM port work (somewhat comfortably) on the PowKiddy V90 (GBA SP-like clamshell) model? Obviously the 'mouse cursor' will be moved using the direction pad. Which games have you tried on the small screen?

_Side-note: I remember playing ScummVM games on the original Nintendo Wii. Quite nice with the WiiMote used as a "mouse". ;-)_

BryMD, 2021-02-15:
Hi Richard! If the PowKiddy runs MiyooCFW, I presume ScummVM will run smoothly :)

The current port basically supports the original roster of 320x240 games (think ScummVM 0.x). Haven't gotten around to adding missing .dat files yet, but assume there's a reason for them being removed from the fork in the first place(?).

TriForceX, 2021-02-15:
> Hi Richard! If the PowKiddy runs MiyooCFW, I presume ScummVM will run smoothly :)
>
> The current port basically supports the original roster of 320x240 games (think ScummVM 0.x). Haven't gotten around [...]

If theres missing .dat files you may find them here https://github.com/scummvm/scummvm/tree/master/dists/engine-data

closed 2021-02-16 by TriForceX