TriForceX/MiyooCFW #104: Bittboy cfw 1.3.1 theme icons

symbios24, 2020-07-25:
Hello,upon trying the latest firmware 1.3.1 on bittboy 3.5 most the icons in all themes except the main theme the system boots first time shown with a gear icon instead their default icon
Is this a bug ??

TriForceX, 2020-07-27:
Maybe the firmware image was installed uncomplete or something is missing on it, because is not a normal thing, most of themes has icons, no one have reported something like that before. i recommend you to reinstall the firmware and try again (remember to format the SD card with SD Card Formatter tool). Or just wait for the next release, which is soon to be ready.

About the smsplusgx, have you tried to put the bios on the emulator folder? "emus/smsplusgx/bios" (not at root in .smsplusgx, that doesnt exist)

symbios24, 2020-07-28:
Strange i did a clean install and used sd fornatter tool but nothing changed all the themes except the main one have messed icons,yes i putted the bios in that folder.i had to create the bios folder coz it wasn't there,any tv out feature on next firmware?

closed 2023-04-20 by Apaczer